ValveSoftware/steamlink-sdk #5: USB script execution as root

TomasDuda, 2016-01-14:
It looks like when you have USB connected to Steam Link when booting, runlevel 1 startup script [is running](https://github.com/ValveSoftware/steamlink-sdk/blob/6a735c5621d7982f45145d729f75698aa71ce724/rootfs/etc/init.d/startup/S01config#L87) the `factory_test/run.sh` script from the USB with root privileges.

Is it really a good idea? It's more like a nice backdoor.

Same for system recovery from `SystemUpdate.zip`.

lahwran, 2016-01-14:
This is dangerous in the case that anyone downloads things onto their usb drive without realizing it - it could be used as a mechanism for an attacker to cross the airgap from a computer to the steam link and take over the steam link. This is also true of automatically enabling ssh. I would recommend requiring that the ssh enablement flag is confirmed in the UI.

mcd1992, 2016-02-03:
If an attacker has physical access to a device then its safe to say that device is no longer 'trusted'.

lahwran, 2016-02-03:
The attacker need not have physical access to execute this attack; all they need to do is convince a user to put a file on a flash drive; this is easily done by telling a user they need to unzip a file in the root of their flash drive. In other words, it doesn't have a warning telling the user that something is probably up and they may be being attacked. Someone who knows what it means will go "nah I know what I'm doing", but if you say "Your flash drive is requesting enabling SSH, which puts your system at risk. You may have a trojan on your flash drive, especially if someone said to ignore this message." then a user who didn't set out to use ssh access will be properly hesitant.

mcd1992, 2016-02-03:
What do you propose as a fix then? A major reason to keep this available at boot is so that we have a method of recovery should something go awry.

The only viable solution I can think of is a recovery/bootloader similar to android that would allow us to enable/disable components or make filesystem modifications prior to booting the kernel. Since the Link has no hardware buttons I guess having the bootloader check for a flag file on a USB stick would suffice. Maybe something like `/steamlink/bootloader_config` that if found would drop us into a simple android like bootloader. Although that would require usb keyboard or controller support in the bootloader which is a lot of overhead. I'm also not sure how much freedom Marvell is giving Valve when dealing with their bootloader.

lahwran, 2016-02-03:
Ah, I see, fair. I suppose I agree there's not much that can be done, then. Too bad though.

closed 2016-09-01 by slouken