ValveSoftware/steamlink-sdk #8: 360 gamepad controls mouse for Metro apps

BlinksTale, 2016-01-15:
Copying this issue over from SteamOS since it sounds like this is a better place for it:

Streaming Windows 10 metro apps will treat a 360 gamepad as though it is mouse input instead. This causes problems for titles like Minecraft Win 10 ed and Xbox One streaming through Steam, though it is still useful for Microsoft Solitaire Collection. It would be great to be able to turn this on and off.

I'd like to try poking through the source code and seeing if I can figure out the problem on my own to help contribute, but aside from checking every file, I'm not sure where to start. Advice on where in the code controllers are handled would help a great deal.

not83, 2016-01-15:
Is this a problem with the steam link sdk

BlinksTale, 2016-01-15:
That's what I'm trying to find out! Streaming is a two side problem: host and client. Steam Link inherently will cover a lot of client end stuff, like controller input, so it depends on both where the controller is turned into mouse input (host vs client) and whether the SDK handles that or if it's even lower level than that.

It might be within the realm of Steam Link's SDK, and I don't have a better place to turn, so worth a shot so long as I'm pursuing this issue.

not83, 2016-01-15:
There is this place

https://steamcommunity.com/groups/homestream/discussions

I think we wait to see what Valve thinks. If they are reading this then
there is no place to report streaming issues

closed 2016-01-15 by slouken

BlinksTale, 2016-01-25:
@not83 I already tried that sadly, the replies were just the community, and not helpful sadly:

https://steamcommunity.com/app/353380/discussions/0/451848855026871092/

Do you have any other ideas on where to go with this?