ValveSoftware/steamlink-sdk #13: Documentation suggestion

bellmounte, 2016-01-16:
I was able to get everything up and running fairly easily, just ran into one easy that doesn't appear to be documented. 

Based on this line I installed Ubuntu 14.04 LTS. 
> Steam Link applications are built using this SDK on Linux.

This does not come with `realpath`, so I had to install it with:
``` bash
apt-get install realpath
```

I would suggest either adding a list of recommended linux distros, or noting that `realpath` is required.

mgerhardy, 2016-01-17:
please see pull request #14

bellmounte, 2016-01-17:
@mgerhardy That solution also works, I'm just not sure what approach the maintainers are interested in.  My assumption is that the Steam Link's distro has `realpath` and so rather than change scripts to match a developer/contributor's machine, I'd suggesting matching the hardware. This assumption is easy enough to test.
Either way, I'd love to hear Valve's thoughts on this issue.

slouken, 2016-01-17:
Pulled #14, thanks!

closed 2016-01-17 by slouken