ValveSoftware/steamlink-sdk #17: Cannot get Kodi to compile

AaronShea, 2016-02-01:
I'm running an Ubuntu 14.04 Server 64-bit install, and I can't seem to get Kodi to compile, even using this automated build script. The process always seems to fail with this error:

```
checking whether the C++ compiler works... no
configure: error: in `/home/aaron/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/JsonSchemaBuilder/native':
configure: error: C++ compiler cannot create executables
See `config.log' for more details
make[1]: *** [native] Error 77
make[1]: Leaving directory `/home/aaron/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/JsonSchemaBuilder'
make: *** [tools/depends/native/JsonSchemaBuilder/bin/JsonSchemaBuilder] Error 2
```

If I just run the script again, I end up with:

```
checking compiler armv7a-cros-linux-gnueabi-gcc --sysroot=/home/aaron/steamlink-sdk/rootfs -marm -mfloat-abi=hard  ... no
configure: error: could not find a working compiler, see config.log for details
make: *** No rule to make target `uninstall'.  Stop.
~/steamlink-sdk/examples/kodi/kodi-src/tools/depends ~/steamlink-sdk/examples/kodi
make: Entering directory `/home/aaron/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native'
Makefile:1: ../Makefile.include: No such file or directory
Makefile:4: *** Error: /share/config.site  is missing. Please reconfigure depends to generate it.  Stop.
make: Leaving directory `/home/aaron/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native'
```

the-kenny, 2016-02-07:
Note that this seems related to upstream changes of Kodi. They seem to have removed some Makefiles and other stuff from `tools/depends/.../`

the-kenny, 2016-02-07:
I'd suggest to freeze the revision of kodi fetched via `examples/kodi/build_steamlink.sh`.

jlnprssnr, 2016-02-10:
@slouken did some adjustments in Commit 84, does that fix your problem?

AaronShea, 2016-02-10:
Still having the same issue, it would be nice to know what commit of Kodi he's building against locally.

```
~/steamlink-sdk/examples/kodi/kodi-src/tools/depends ~/steamlink-sdk/examples/kodi
~/steamlink-sdk/examples/kodi
~/steamlink-sdk/examples/kodi/kodi-src/tools/depends ~/steamlink-sdk/examples/kodi
checking build system type... armv7l-unknown-linux-gnueabihf
checking host system type... armv7a-cros-linux-gnueabi
checking whether the C++ compiler works... no
configure: error: in `/home/aaron/steamlink-sdk/examples/kodi/kodi-src/tools/depends':
configure: error: C++ compiler cannot create executables
See `config.log' for more details
checking build system type... armv7l-unknown-linux-gnueabihf
checking host system type... armv7a-cros-linux-gnueabi
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for armv7a-cros-linux-gnueabi-strip... armv7a-cros-linux-gnueabi-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking ABI=standard
checking compiler armv7a-cros-linux-gnueabi-gcc --sysroot=/home/aaron/steamlink-sdk/rootfs -marm -mfloat-abi=hard  ... no
configure: error: could not find a working compiler, see config.log for details
make: *** No rule to make target `uninstall'.  Stop.
~/steamlink-sdk/examples/kodi/kodi-src/tools/depends ~/steamlink-sdk/examples/kodi
make: Entering directory `/home/aaron/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native'
Makefile:1: ../Makefile.include: No such file or directory
Makefile:4: *** Error: /share/config.site  is missing. Please reconfigure depends to generate it.  Stop.
make: Leaving directory `/home/aaron/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native'
```

slouken, 2016-02-10:
I'm building against master as of last night. What do you see in the config.log?

AaronShea, 2016-02-11:
Here's `config.log` in `kodi-src/tools/depends`

```
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by xbmc-depends configure 2.00, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ./configure --with-toolchain=/home/aaron/steamlink-sdk/toolchain --prefix=/home/aaron/steamlink-sdk/kodi-deps --host=armv7a-cros-linux-gnueabi

## --------- ##
## Platform. ##
## --------- ##

hostname = ubuntu
uname -m = armv7l
uname -r = 3.8.13-mrvl
uname -s = Linux
uname -v = #35 PREEMPT Thu Jan 21 11:20:45 PST 2016

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /home/aaron/steamlink-sdk/toolchain/bin
PATH: /home/aaron/steamlink-sdk/bin
PATH: /home/aaron/steamlink-sdk/external/qt-everywhere-opensource-src-5.4.1/build/host/bin
PATH: /home/aaron/steamlink-sdk/toolchain/bin
PATH: /home/aaron/steamlink-sdk/bin
PATH: /home/aaron/steamlink-sdk/external/qt-everywhere-opensource-src-5.4.1/build/host/bin
PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin
PATH: /usr/games
PATH: /usr/local/games

## ----------- ##
## Core tests. ##
## ----------- ##

configure:1993: checking build system type
configure:2007: result: armv7l-unknown-linux-gnueabihf
configure:2027: checking host system type
configure:2040: result: armv7a-cros-linux-gnueabi
configure:2216: checking for C++ compiler version
configure:2225: armv7a-cros-linux-gnueabi-g++ --sysroot=/home/aaron/steamlink-sdk/rootfs -marm -mfloat-abi=hard --version >&5
armv7a-cros-linux-gnueabi-g++.real.elf (4.9.1_cos_gg_2f9796c_4.9.1-r86) 4.9.x-google 20140827 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

configure:2236: $? = 0
configure:2225: armv7a-cros-linux-gnueabi-g++ --sysroot=/home/aaron/steamlink-sdk/rootfs -marm -mfloat-abi=hard -v >&5
Using built-in specs.

COLLECT_GCC=/home/aaron/steamlink-sdk/toolchain/bin/../usr/x86_64-pc-linux-gnu/armv7a-cros-linux-gnueabi/gcc-bin/4.9.x-google/armv7a-cros-linux-gnueabi-g++.real.elf
COLLECT_LTO_WRAPPER=/home/aaron/steamlink-sdk/toolchain/usr/x86_64-pc-linux-gnu/armv7a-cros-linux-gnueabi/gcc-bin/4.9.x-google/../../../../libexec/gcc/armv7a-cros-linux-gnueabi/4.9.x-google/lto-wrapper
Target: armv7a-cros-linux-gnueabi
Configured with: /var/tmp/portage/cross-armv7a-cros-linux-gnueabi/gcc-4.9.1-r86/work/gcc-4.9.1/configure --disable-multilib --prefix=/usr --bindir=/usr/x86_64-pc-linux-gnu/armv7a-cros-linux-gnueabi/gcc-bin/4.9.x-google --datadir=/usr/share/gcc-data/armv7a-cros-linux-gnueabi/4.9.x-google --disable-canonical-system-headers --mandir=/usr/share/gcc-data/armv7a-cros-linux-gnueabi/4.9.x-google/man --infodir=/usr/share/gcc-data/armv7a-cros-linux-gnueabi/4.9.x-google/info --includedir=/usr/lib/gcc/armv7a-cros-linux-gnueabi/4.9.x-google/include --with-gxx-include-dir=/usr/lib/gcc/armv7a-cros-linux-gnueabi/4.9.x-google/include/g++-v4 --enable-libatomic --with-python-dir=/share/gcc-data/armv7a-cros-linux-gnueabi/4.9.x-google/python --host=x86_64-pc-linux-gnu --target=armv7a-cros-linux-gnueabi --build=x86_64-pc-linux-gnu --enable-languages=c,c++,go --with-float=hard --with-mode=thumb --with-sysroot=/usr/armv7a-cros-linux-gnueabi --disable-libmudflap --disable-libssp --enable-libgomp --enable-__cxa_atexit --enable-checking=release --disable-libquadmath --disable-libitm --disable-libcilkrts --with-arch=armv7-a --disable-esp --with-bugurl=http://code.google.com/p/chromium-os/issues/entry --with-pkgversion=4.9.1_cos_gg_2f9796c_4.9.1-r86 --enable-linker-build-id
[... 177 lines cut ...]
use_cpu=''
use_firmware=''
use_gplv3=''
use_host=''
use_ndk=''
use_sdk=''
use_sdk_path=''
use_tarballs=''
use_toolchain=''
use_xcode=''

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "xbmc-depends"
#define PACKAGE_TARNAME "xbmc-depends"
#define PACKAGE_VERSION "2.00"
#define PACKAGE_STRING "xbmc-depends 2.00"
#define PACKAGE_BUGREPORT "http://trac.xbmc.org"
#define PACKAGE_URL ""

configure: exit 77
```

slouken, 2016-02-11:
Okay, I just pushed a change that I think will fix this. Can you confirm?

AaronShea, 2016-02-12:
Seems to compile just fine now, I did have to make one small modification though. Seems there can be a conflict between `sys/xattr.h` and `linux/xattr.h` when compiling the Samba library. This is more on Kodi and Samba though. To fix it I just removed `sys/capability.h` from the offending includes.h file

slouken, 2016-02-12:
This conflict should have been fixed here:
https://github.com/ValveSoftware/steamlink-sdk/commit/3c89376abde5b9a7f67f19cc3c83b4ef0c4a86b1

Are you fully up to date?

AaronShea, 2016-02-13:
Fresh pull from the repo and newest packages on my host system. Still seems to run into that issue, which is odd.

closed 2016-09-01 by slouken

vaidyk, 2017-03-07:
Just ran into the same issue on 16.04.  Installed Python to get past it.