ValveSoftware/steamlink-sdk #18: Having trouble compiling RetroArch

ProfessorKaos64, 2016-02-03:
Regardless of the entire thing working or not, I want to find out why the C compiler is reported as broken. A [particular script](https://raw.githubusercontent.com/libretro/RetroArch/e84b860bf61f6bc276a344d2defd8b1ce9f17ef2/qb/qb.comp.sh) looks for a C compiler by performing:

```
cc_works=0
if [ "$CC" ]; then
	"$CC" -o "$TEMP_EXE" "$TEMP_C" >/dev/null 2>&1 && cc_works=1
else
	for CC in ${CC:=$(which ${CROSS_COMPILE}gcc ${CROSS_COMPILE}cc ${CROSS_COMPILE}clang 2>/dev/null)} ''; do
		"$CC" -o "$TEMP_EXE" "$TEMP_C" >/dev/null 2>&1 && cc_works=1 && break
	done
fi

rm -f "$TEMP_C" "$TEMP_EXE"

cc_status='does not work'
if [ "$cc_works" = '1' ]; then
	cc_status='works'
elif [ -z "$CC" ]; then
	cc_status='not found'
fi

echo "Checking for suitable working C compiler ... $CC $cc_status"

if [ "$cc_works" = '0' ] && [ "$USE_LANG_C" = 'yes' ]; then
	echo "Error: Cannot proceed without a working C compiler."
	exit 1
fi
```

When running the configure statements (sourced from [raspberry pi forums](https://www.raspberrypi.org/forums/viewtopic.php?t=56070)), I get:

```
Checking for suitable working C compiler ... armv7a-cros-linux-gnueabi-clang --sysroot=$MARVELL_ROOTFS -marm -mfloat-abi=hard does not work
```

How can I set the clang compiler? Any hints?

**Links**  
* [My script](https://github.com/ProfessorKaos64/steamlink-sdk/raw/retroarch/examples/retroarch/build_steamlink.sh)

slouken, 2016-02-03:
It looks like having "$CC" in quotes is causing the problem. Try just using $CC

ProfessorKaos64, 2016-02-03:
Hi @slouken , tried the following to no avail:

```
cc_works=0
if [ $CC ]; then
	$CC -o "$TEMP_EXE" "$TEMP_C" && cc_works=1
else
	for CC in ${CC:=$(which ${CROSS_COMPILE}gcc ${CROSS_COMPILE}cc ${CROSS_COMPILE}clang)} ''; do
		$CC -o "$TEMP_EXE" "$TEMP_C" && cc_works=1 && break
	done
fi

rm -f "$TEMP_C" "$TEMP_EXE"

cc_status='does not work'
if [ "$cc_works" = '1' ]; then
	cc_status='works'
elif [ -z $CC ]; then
	cc_status='not found'
fi

echo "Checking for suitable working C compiler ... $CC $cc_status"

if [ "$cc_works" = '0' ] && [ "$USE_LANG_C" = 'yes' ]; then
	echo "Error: Cannot proceed without a working C compiler."
	exit 1
fi
```

Should I only remove the quotes on the `-z $CC` test? If it's a globbing issue, I wasn't sure if all or just that test needed patched.

ProfessorKaos64, 2016-02-03:
I think I got it, now C++ complains, I'll fix it the same way and see where I get.

ProfessorKaos64, 2016-02-03:
I see windres may be possible via one of the external packages, any idea which one may work? Seems like SDL2. Sorry for being a bit new at this, I do Debian packaging, but i'm new to ARM work.

closed 2016-02-03 by ProfessorKaos64