ValveSoftware/steamlink-sdk #22: Cave Express: garbled font graphics

angelosa, 2016-03-05:
I've tried to run Cave Express as a deploy test to Steam Link and all of the menu text is garbled. 
Happens with the cavepacker extension too.

[image]

mgerhardy, 2016-03-06:
Thanks for the report - which revision of the caveexpress repository are you on? I recently switched to a font that also supports unicode.

Most likely related to https://github.com/mgerhardy/caveexpress/issues/85

Usually it helps if you delete base/$game/textures/complete.lua and rerun cmake (resp. the steamlink.sh script).

And for the sdk - maybe we should update the script to checkout a specific tag - e.g. latest tag is 2.4 (https://github.com/mgerhardy/caveexpress/tree/2.4)

closed 2016-03-09 by slouken

mgerhardy, 2016-03-11:
Looks like this is a problem with the small texture set - run ./cavepacker -set texturedefintiion big to solve it, too. Will continue to document the stuff in the linked ticket.

mgerhardy, 2016-03-20:
btw: fixed in master