ValveSoftware/steamlink-sdk #26: Kodi patch now out of date, multiple issues

ezelkow1, 2016-06-02:
I was trying to build kodi but it looks like its very out of date now. So before I dug deeper I figured Id post what I found so far, these are the changes Ive done to the patch to get it this far:

--- kodi.old/configure.ac       2016-01-31 01:08:43.988534081 -0800
+++ kodi/configure.ac   2016-01-31 01:09:16.353783633 -0800
@@ +520,7 @@
      target_platform=target_linux
      ARCH="arm"
      use_arch="arm"
+     #use_joystick=no
      use_neon=yes
      use_gles=yes
      use_gl=no
@@ -1659,7 +1659,7 @@

******************************use joystick no longer exists, also removed the block below that does a usejoystick test*********************

@@ -36,7 +36,7 @@
-       cd $(PLATFORM); AR="$(AR)" CFLAGS="$(CFLAGS)" CC=$(CC) RANLIB=$(RANLIB) $(CONFIGURE)
+       cd $(PLATFORM); AR="$(AR)" CFLAGS="$(CFLAGS)" CC="$(CC)" RANLIB=$(RANLIB) $(CONFIGURE)
        if test "$(OS)" = "osx"; then \
                sed -ie "s|CC= /usr/bin/gcc-4.2|CC= $(CC)|" "$(PLATFORM)/Makefile"; \
                sed -ie "s|CFLAG= |CFLAG=$(CFLAGS) |" "$(PLATFORM)/Makefile"; \

****************they removed the block comment above this, so the patch check no longer works *********

And now the part thats probably going to get complicated:
patch: **** malformed patch at line 457: diff -ruN -x .git kodi.old/xbmc/windowing/egl/EGLNativeTypeSDL.cpp kodi/xbmc/windowing/egl/EGLNativeTypeSDL.cpp

It looks like they've completely removed SDL as that file no longer exists

ezelkow1, 2016-06-03:
[kodi.patch.txt](https://github.com/ValveSoftware/steamlink-sdk/files/297865/kodi.patch.txt)
attaching my current patch as a txt since thats all github supports

slouken, 2016-07-14:
Updated!

closed 2016-07-14 by slouken