ValveSoftware/steamlink-sdk #31: Sound does not work in kodi example

LeBlue, 2016-07-06:
I managed to manually compile the kodi example and get it running on the steamlink (build 518), sdk revision 59d94cc:

Video output is working, but I don't get any sound, neither UI sounds nor when playing video or audio files. Video and sound output are connected via HDMI.

Things I checked:
- Pulseaudio server is running and kodi successfully connects to it.
- Kodi is linked to pulse-8 and the libraries are present on the device.
- Sound is working for the steamlink shell application, as long as pulseaudio is running.

Possible reasons:
- Pulseaudio does not seem to recognize the HDMI audio output, it only lists the nullsink as default.
- Pulseaudio does not  have passthrough capabilities, which kodi seems to require.

pulseaudio -vvvv output:
```
W: [pulseaudio] main.c: This program is not intended to be run as root (unless --system is specified).
D: [pulseaudio] core-rtclock.c: Timer slack is set to 50 us.
D: [pulseaudio] core-util.c: setpriority() worked.
I: [pulseaudio] core-util.c: Successfully gained nice level -11.
I: [pulseaudio] main.c: This is PulseAudio 8.0
D: [pulseaudio] main.c: Compilation host: armv7a-cros-linux-gnueabi
D: [pulseaudio] main.c: Compilation CFLAGS: -g -O2 -Wall -W -Wextra -pipe -Wno-long-long -Wno-overlength-strings -Wunsafe-loop-optimizations -Wundef -Wformat=2 -Wlogical-op -
Wsign-compare -Wformat-security -Wmissing-include-dirs -Wformat-nonliteral -Wpointer-arith -Winit-self -Wdeclaration-after-statement -Wfloat-equal -Wmissing-prototypes -Wredu
ndant-decls -Wmissing-declarations -Wmissing-noreturn -Wshadow -Wendif-labels -Wcast-align -Wstrict-aliasing -Wwrite-strings -Wno-unused-parameter -ffast-math -fno-common -fd
iagnostics-show-option -fdiagnostics-color=auto
D: [pulseaudio] main.c: Running on host: Linux armv7l 3.8.13-mrvl #45 PREEMPT Tue Jun 7 13:49:32 PDT 2016
D: [pulseaudio] main.c: Found 1 CPUs.
I: [pulseaudio] main.c: Page size is 4096 bytes
D: [pulseaudio] main.c: Compiled with Valgrind support: no
D: [pulseaudio] main.c: Running in valgrind mode: no
D: [pulseaudio] main.c: Running in VM: no
D: [pulseaudio] main.c: Optimized build: yes
D: [pulseaudio] main.c: FASTPATH defined, only fast path asserts disabled.
I: [pulseaudio] main.c: Machine ID is 86cb1be968ebd800cf30367e00000001.
I: [pulseaudio] main.c: Using runtime directory /home/steam/.config/pulse/86cb1be968ebd800cf30367e00000001-runtime.
I: [pulseaudio] main.c: Using state directory /home/steam/.config/pulse.
I: [pulseaudio] main.c: Using modules directory /usr/lib/pulse-8.0/modules.
I: [pulseaudio] main.c: Running in system mode: no
I: [pulseaudio] main.c: System appears to not support high resolution timers
D: [pulseaudio] memblock.c: Using shared memory pool with 1024 slots of size 64.0 KiB each, total size is 64.0 MiB, maximum usable slot size is 65496
D: [pulseaudio] memblock.c: Using shared memory pool with 1024 slots of size 64.0 KiB each, total size is 64.0 MiB, maximum usable slot size is 65496
I: [pulseaudio] cpu-arm.c: CPU flags: V6 V7 VFP EDSP NEON VFPV3
I: [pulseaudio] svolume_arm.c: Initialising ARM optimized volume functions.
I: [pulseaudio] sconv_neon.c: Initialising ARM NEON optimized conversions.
I: [pulseaudio] mix_neon.c: Initialising ARM NEON optimized mixing functions.
I: [pulseaudio] remap_neon.c: Initialising ARM NEON optimized remappers.
I: [pulseaudio] module-device-restore.c: Successfully opened database file '/home/steam/.config/pulse/86cb1be968ebd800cf30367e00000001-device-volumes'.
I: [pulseaudio] module.c: Loaded "module-device-restore" (index: #0; argument: "").
I: [pulseaudio] module-stream-restore.c: Successfully opened database file '/home/steam/.config/pulse/86cb1be968ebd800cf30367e00000001-stream-volumes'.
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1 added for object /org/pulseaudio/stream_restore1
I: [pulseaudio] module.c: Loaded "module-stream-restore" (index: #1; argument: "").
I: [pulseaudio] module-card-restore.c: Successfully opened database file '/home/steam/.config/pulse/86cb1be968ebd800cf30367e00000001-card-database'.
I: [pulseaudio] module.c: Loaded "module-card-restore" (index: #2; argument: "").
I: [pulseaudio] module.c: Loaded "module-augment-properties" (index: #3; argument: "").
I: [pulseaudio] module.c: Loaded "module-switch-on-port-available" (index: #4; argument: "").
D: [pulseaudio] module.c: Checking for existence of '/usr/lib/pulse-8.0/modules/module-udev-detect.so': success
I: [pulseaudio] module-udev-detect.c: Found 0 cards.
I: [pulseaudio] module.c: Loaded "module-udev-detect" (index: #5; argument: "tsched=0").
D: [pulseaudio] module.c: Checking for existence of '/usr/lib/pulse-8.0/modules/module-jackdbus-detect.so': failure
D: [pulseaudio] module.c: Checking for existence of '/usr/lib/pulse-8.0/modules/module-dbus-protocol.so': success
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Memstats added for object /org/pulseaudio/core1/memstats
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Module added for object /org/pulseaudio/core1/module0
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Module added for object /org/pulseaudio/core1/module1
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Module added for object /org/pulseaudio/core1/module2
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Module added for object /org/pulseaudio/core1/module3
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Module added for object /org/pulseaudio/core1/module4
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Module added for object /org/pulseaudio/core1/module5
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Module added for object /org/pulseaudio/core1/module6
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1 added for object /org/pulseaudio/core1
I: [pulseaudio] module.c: Loaded "module-dbus-protocol" (index: #6; argument: "").
[... 192 lines cut ...]
00:14:05 T:2990314464    INFO: CActiveAESink::OpenSink - initialize sink
00:14:05 T:2990314464   DEBUG: CActiveAESink::OpenSink - trying to open device PULSE:Default
00:14:05 T:2990314464   DEBUG: PulseAudio: Context authorizing
00:14:05 T:2990314464   DEBUG: PulseAudio: Context setting name
00:14:05 T:2990314464   DEBUG: PulseAudio: Context ready
00:14:05 T:2990314464   ERROR: PulseAudio: Sink Default not found
00:14:05 T:2990314464   DEBUG: CActiveAESink::OpenSink - trying to open device PULSE:Default
00:14:05 T:2990314464   DEBUG: PulseAudio: Context authorizing
00:14:05 T:2990314464   DEBUG: PulseAudio: Context setting name
00:14:05 T:2990314464   DEBUG: PulseAudio: Context ready
00:14:05 T:2990314464   ERROR: PulseAudio: Sink Default not found
00:14:05 T:2990314464   DEBUG: CActiveAESink::OpenSink - open NULL sink
00:14:05 T:2981925856   DEBUG: Thread AESinkNull start, auto delete: false
00:14:05 T:2981925856   DEBUG: CAESinkNULL::Process
00:14:05 T:2990314464   DEBUG: CActiveAESink::OpenSink - NULL Initialized:
00:14:05 T:2990314464   DEBUG:   Output Device : Default
00:14:05 T:2990314464   DEBUG:   Sample Rate   : 44100
00:14:05 T:2990314464   DEBUG:   Sample Format : AE_FMT_FLOAT
00:14:05 T:2990314464   DEBUG:   Channel Count : 2
00:14:05 T:2990314464   DEBUG:   Channel Layout: FL,FR
00:14:05 T:2990314464   DEBUG:   Frames        : 11000
00:14:05 T:2990314464   DEBUG:   Frame Size    : 8
00:14:05 T:2924471264 WARNING: ActiveAE::Configure - sink returned large buffer of 249 ms, reducing to 100 ms
....
```

sWski, 2016-07-08:
I have the same setup and audio is not working as well. Did you managed to play some audio file from shell? I tried to play some .wav files with `paplay`/`pacat` commands and it always failed with some bad api call to `libsndfile`. The Link app must be using different way, becase HDMI audio output is working there... :disappointed:

slouken, 2016-07-14:
Should be fixed with the latest commits. Thanks!

closed 2016-07-14 by slouken