ValveSoftware/steamlink-sdk #38: Kodi built archive extraction failed, symlink error

kroppt, 2016-10-26:
After building, I get the following file structure: `./steamlink/apps/kodi.tgz`. This alone does not copy into the SteamLink.

When extracting the archive on Linux, I get the following error: `tar: kodi/include/xmbc: Cannot create symlink to 'kodi': Operation not permitted`. On Windows, it seems to think that the file doesn't exist anymore. This happens for a few files in the archive.

When ignoring these errors and putting Kodi onto the SteamLink, the app boots into a black screen for a few seconds, then quits back to the main SteamLink menu.

I saw on on of the guides for another app (mame4all) that USB drives had to be FAT32, but this did not resolve the issue with extracting the archive or installing Kodi onto the SteamLink.

jonesmz, 2016-12-25:
@kroppt Could you try again? Several patches were just merged for Kodi

slouken, 2016-12-27:
I think this is taken care of in the new Kodi build patches

closed 2016-12-27 by slouken