ValveSoftware/steamlink-sdk #39: Directly send D3D11 / OpenGL texture/render buffers to steam link?

zezba9000, 2016-11-07:
Is it possible to directly send D3D11/D3D12/GL texture/render buffers to a steam link API? Is there an SDK that would allow for this?

We would like to take advantage of the low latency compression tech without the overhead of the Windows compositor. Directly sending a RenderTarget/Texture to a SteamLink API would fix the lag we see caused by Windows.

Sorry if this is bad place to ask but I'm not sure if there is an API that supports this?

zezba9000, 2016-11-08:
In case I wasn't very clear as I understand it Steam Link is compressing screen buffers of the Windows desktop. This is slow as it must wait for the Windows compositor. I want to directly send the Steam Link API a buffer directly to bypass the Windows compositor.

slouken, 2016-12-02:
There isn't an API to do this, however Steam generally reads the game's backbuffer before the present, avoiding the overhead of the Windows compositor.

closed 2016-12-02 by slouken

zezba9000, 2016-12-02:
So if it does work as you say (directly accessing the games DXGI backbuffer thats created in a swapchain and NOT the compositors triple buffered backbuffer that copies your swapchains texture) then turning off vSync in the game should yield better results correct?

Because we don't see better results it would appear its grabbing the compositors backbuffer DXGI object. Am I missing something?