ValveSoftware/steamlink-sdk #41: Kodi build fails

parkerlreed, 2016-11-13:
Arch x86_64
As per the instructions all I ran was the build script inside the Kodi folder.

````
make[5]: Entering directory '/home/parker/build/steamlink-sdk/examples/kodi/kodi-src/tools/depends/target/libvorbis/armv7a-cros-linux-gnueabi/lib'
/bin/sh ../libtool --tag=CC   --mode=link /home/parker/build/steamlink-sdk/toolchain/bin/armv7a-cros-linux-gnueabi-gcc --sysroot=/home/parker/build/steamlink-sdk/rootfs -marm -mfloat-abi=hard  -O20 -Wall -Wextra -ffast-math -D_REENTRANT -fsigned-char -Wdeclaration-after-statement    -isystem /home/parker/build/steamlink-sdk/kodi-deps/armv7a-cros-linux-gnueabi/include  -DUSE_MEMORY_H -no-undefined -version-info 4:5:4 -L/home/parker/build/steamlink-sdk/kodi-deps/armv7a-cros-linux-gnueabi/lib  -static-libgcc -static-libstdc++   -o libvorbis.la -rpath /home/parker/build/steamlink-sdk/kodi-deps/armv7a-cros-linux-gnueabi/lib mdct.lo smallft.lo block.lo envelope.lo window.lo lsp.lo lpc.lo analysis.lo synthesis.lo psy.lo info.lo floor1.lo floor0.lo res0.lo mapping0.lo registry.lo codebook.lo sharedbook.lo lookup.lo bitrate.lo -lm -L/home/parker/build/steamlink-sdk/kodi-deps/armv7a-cros-linux-gnueabi/lib -logg -lm
libtool: link: /home/parker/build/steamlink-sdk/toolchain/bin/armv7a-cros-linux-gnueabi-gcc --sysroot=/home/parker/build/steamlink-sdk/rootfs -marm -mfloat-abi=hard -shared  .libs/mdct.o .libs/smallft.o .libs/block.o .libs/envelope.o .libs/window.o .libs/lsp.o .libs/lpc.o .libs/analysis.o .libs/synthesis.o .libs/psy.o .libs/info.o .libs/floor1.o .libs/floor0.o .libs/res0.o .libs/mapping0.o .libs/registry.o .libs/codebook.o .libs/sharedbook.o .libs/lookup.o .libs/bitrate.o   -L/home/parker/build/steamlink-sdk/kodi-deps/armv7a-cros-linux-gnueabi/lib /usr/lib/libogg.so -lm  -marm -mfloat-abi=hard   -Wl,-soname -Wl,libvorbis.so.0 -o .libs/libvorbis.so.0.4.5
/usr/lib/libogg.so: file not recognized: File format not recognized
collect2.elf: error: ld returned 1 exit status
make[5]: *** [Makefile:369: libvorbis.la] Error 1
make[5]: Leaving directory '/home/parker/build/steamlink-sdk/examples/kodi/kodi-src/tools/depends/target/libvorbis/armv7a-cros-linux-gnueabi/lib'
make[4]: *** [Makefile:475: all-recursive] Error 1
make[4]: Leaving directory '/home/parker/build/steamlink-sdk/examples/kodi/kodi-src/tools/depends/target/libvorbis/armv7a-cros-linux-gnueabi/lib'
make[3]: *** [Makefile:371: all-recursive] Error 1
make[3]: Leaving directory '/home/parker/build/steamlink-sdk/examples/kodi/kodi-src/tools/depends/target/libvorbis/armv7a-cros-linux-gnueabi'
make[2]: *** [Makefile:253: all] Error 2
make[2]: Leaving directory '/home/parker/build/steamlink-sdk/examples/kodi/kodi-src/tools/depends/target/libvorbis/armv7a-cros-linux-gnueabi'
make[1]: *** [Makefile:31: armv7a-cros-linux-gnueabi/lib/.libs/libvorbis.a] Error 2
make[1]: Leaving directory '/home/parker/build/steamlink-sdk/examples/kodi/kodi-src/tools/depends/target/libvorbis'
make: *** [Makefile:109: libvorbis] Error 2
make: Leaving directory '/home/parker/build/steamlink-sdk/examples/kodi/kodi-src/tools/depends/target'
````

mtdeguzis, 2016-12-01:
I actually get this error on Debian Jessie:

```

amiga23, 2016-12-11:
@parkerlreed did you solve the issue meanwhile? I do have the same issue.
I do not know how to solve but I do have a workarround:

1. cd into the directory where the uild failed
2. run the linker manualy and replace -logg with the absolute path the correct libogg.so
3. repeat this for all the following errors related to libogg

So in your case to solve the first:
`/home/parker/build/steamlink-sdk/examples/kodi/kodi-src/tools/depends/target/libvorbis/armv7a-cros-linux-gnueabi/lib
bin/sh ../libtool --tag=CC   --mode=link /home/parker/build/steamlink-sdk/toolchain/bin/armv7a-cros-linux-gnueabi-gcc --sysroot=/home/parker/build/steamlink-sdk/rootfs -marm -mfloat-abi=hard  -O20 -Wall -Wextra -ffast-math -D_REENTRANT -fsigned-char -Wdeclaration-after-statement    -isystem /home/parker/build/steamlink-sdk/kodi-deps/armv7a-cros-linux-gnueabi/include  -DUSE_MEMORY_H -no-undefined -version-info 4:5:4 -L/home/parker/build/steamlink-sdk/kodi-deps/armv7a-cros-linux-gnueabi/lib  -static-libgcc -static-libstdc++   -o libvorbis.la -rpath /home/parker/build/steamlink-sdk/kodi-deps/armv7a-cros-linux-gnueabi/lib mdct.lo smallft.lo block.lo envelope.lo window.lo lsp.lo lpc.lo analysis.lo synthesis.lo psy.lo info.lo floor1.lo floor0.lo res0.lo mapping0.lo registry.lo codebook.lo sharedbook.lo lookup.lo bitrate.lo -lm -L/home/parker/build/steamlink-sdk/kodi-deps/armv7a-cros-linux-gnueabi/lib /home/parker/build/steamlink-sdk/rootfs/usr/lib/libogg.so -lm
`

I tried setting -L/home/parker/build/steamlink-sdk/rootfs/usr/lib but that did not help.

@mdeguzis your problem is different, maybe you are using a wrong filesystemon /mnt/bulk-storage/ ?

parkerlreed, 2016-12-11:
My temp solution was to symlink my system logg to the SDK version. I haven't tried a recent compile to see if anything changed.

parkerlreed, 2016-12-20:
Tried compiling again and it again picks ups the system libogg.

jonesmz, 2016-12-25:
@parkerlreed Could you try again? Several patches for Kodi were just merged today.

parkerlreed, 2016-12-26:
@jonesmz I will check as soon as the internet stops being crappy at my grandparents...

````
Receiving objects:  31% (84388/270087), 247.61 MiB | 307.00 KiB/s   
````

Happy holidays to all!

parkerlreed, 2016-12-26:
Umm heads up, git am requires you to login?

````
[parker@satellite kodi]$ ./build_steamlink.sh 
Cloning into '/home/parker/build/steamlink-sdk/examples/kodi/kodi-src'...
remote: Counting objects: 412850, done.
remote: Compressing objects: 100% (22/22), done.
remote: Total 412850 (delta 5), reused 0 (delta 0), pack-reused 412828
Receiving objects: 100% (412850/412850), 567.08 MiB | 425.00 KiB/s, done.
Resolving deltas: 100% (299850/299850), done.
~/build/steamlink-sdk/examples/kodi/kodi-src ~/build/steamlink-sdk/examples/kodi

*** Please tell me who you are.

Run

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

to set your account's default identity.
Omit --global to set the identity only in this repository.

fatal: unable to auto-detect email address (got 'parker@satellite.(none)')
````

It fails on this line https://github.com/ValveSoftware/steamlink-sdk/blob/master/examples/kodi/build_steamlink.sh#L23

parkerlreed, 2016-12-26:
Is this supposed to be using my native gcc to build python?

````
/usr/bin/gcc -pthread -c -fno-strict-aliasing -I/home/parker/build/steamlink-sdk/kodi-deps/armv7l-linux-gnueabihf-native/include  -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes  -I. -IInclude -I./Include -I/home/parker/build/steamlink-sdk/kodi-deps/armv7l-linux-gnueabihf-native/include   -DPy_BUILD_CORE -o Objects/unicodectype.o Objects/unicodectype.c
````

parkerlreed, 2016-12-26:
As per the original issue, everything is good now. Built fine and starts on the Link. 

I've noticed the Steam Controller doesn't seem to respond at all in Krypton but I'm sure that's a simple other issue.

closed 2016-12-26 by parkerlreed