ValveSoftware/steamlink-sdk #46: How to create custom controller mappings for the Steam Link

slouken, 2016-11-30:
Grab the Steam Link SDK from here:
https://github.com/ValveSoftware/steamlink-sdk

Then grab the source to SDL from here:
http://www.libsdl.org/tmp/SDL-2.0.zip

Build the controllermap test program and copy it and the BMP files from the test directory to the Steam Link.

Then kill the Steam Link shell:
killall app_run.sh shell

Then plug in your controller and run controllermap 0 to bring up the mapping UI and create a mapping string.

Post the mapping string so we can add it to the official firmware, and you can use it locally by editing /home/steam/rc.local and adding the line
export SDL_GAMECONTROLLERCONFIG="<insert line here>"
where the other environment variables are set.

If you have multiple controllers, you should be able to add multiple mappings, separated by newlines, to the environment variable.

Cheers!

parkerlreed, 2016-12-27:
Couldn't this be closed/updated now that there is native remapping built-in that saves to the same file?

slouken, 2016-12-27:
Yup!

closed 2016-12-27 by slouken