ValveSoftware/steamlink-sdk #48: Other resolutions / aspects other then 16:9?

zezba9000, 2016-12-02:
Is it possible for the SteamLink OS / software to be updated to support other resolutions like 1440p or 800p etc?

It would also be cool if it supported portrait screens that rotates a landscape source. Are these ideas possible with the hardware / software?

slouken, 2017-01-07:
These are not supported HDMI resolutions, sorry!

zezba9000, 2017-01-07:
I'm not sure what you mean as my Asus Chromebit can run at 1280x768 over HDMI on a 1280x768 TV.
Are you saying the hardware doesn't support it on the steamlink device?

The RaspberryPi also supports this output over HDMI.

slouken, 2017-01-07:
My understanding is that the hardware doesn't support them, but I'll check with Marvell and confirm.

Thanks for checking!

slouken, 2017-01-08:
Can you add a link to the debug logs so we can get the EDID information from your TV?
Go to the System menu on the Steam Link, click the "Save debug logs" button, then start streaming, and then look for logs\streaming_log.txt from the Steam folder on your PC.

zezba9000, 2017-01-08:
I can just bin dump the EDID from Linux instead if that works?

slouken, 2017-01-08:
Sure, that would work

greyltc, 2017-03-15:
I'm looking for some more flexibility in resolution setting as well.

Here's my decoded EDID:
```
Section "Monitor"
	Identifier "Acer AL2671W"
	ModelName "Acer AL2671W"
	VendorName "ACR"
	# Monitor Manufactured week 39 of 2005
	# EDID version 1.3
	# Digital Display
	DisplaySize 570 340
	Gamma 2.20
	Option "DPMS" "true"
	Horizsync 30-80
	VertRefresh 50-85
	# Maximum pixel clock is 80MHz
	#Not giving standard mode: 640x480, 85Hz
	#Not giving standard mode: 800x600, 85Hz
	#Not giving standard mode: 1024x768, 85Hz
	#Not giving standard mode: 248x155, 60Hz
	#Not giving standard mode: 248x155, 60Hz
	#Not giving standard mode: 248x155, 60Hz
	#Not giving standard mode: 248x155, 60Hz
	#Not giving standard mode: 248x155, 60Hz
	Modeline 	"Mode 0" 79.50 1280 1344 1472 1664 768 771 778 798 +hsync +vsync 
EndSection
```

This display has a native resolution of 1280x768 and I'd like to drive it at that.

raw edid bytes attached: [edid.bin.zip](https://github.com/ValveSoftware/steamlink-sdk/files/843149/edid.bin.zip)

Manouchehri, 2018-01-26:
Since there's no xrandr, how can we change the resolution from the terminal?

MichaelDePiazzi, 2018-02-09:
@slouken Any update on whether this might be possible?

My TV is also 768p (1024x768 @ 60Hz / 16:9 aspect ratio), and I'd love to be able to set my Steam Link to output at the native resolution.

My Raspberry Pi running OSMC (Kodi) has no issue setting my TV to it's 1024x768 60Hz native resolution.

If it helps, here is my TV's EDID dump from the Steam Link logs:
```
 ------------- block 0 -----------
00 ff ff ff ff ff ff 00 42 49 00 16 01 00 00 00 ,
2d 11 01 03 81 46 27 78 0a de 58 a3 54 4c 99 26 ,
0f 45 47 af cf 00 81 40 81 8f d1 c0 01 01 01 01 ,
01 01 81 80 01 01 64 19 00 40 41 00 26 30 18 88 ,
36 00 c4 8e 21 00 00 18 01 1d 80 18 71 1c 16 20 ,
58 2c 25 00 c4 8e 21 00 00 9e 00 00 00 fc 00 50 ,
72 69 6d 61 20 54 56 0a 20 20 20 20 00 00 00 fd ,
00 30 4c 0f 50 11 00 0a 20 20 20 20 20 20 01 11 

 ------------- block 1 -----------
02 03 25 76 4e 02 03 04 05 06 07 11 12 13 14 15 ,
16 90 9f 26 09 17 07 17 17 50 83 01 00 00 66 03 ,
0c 00 20 00 80 66 21 50 b0 51 00 1b 30 40 70 36 ,
00 c4 8e 21 00 00 1e 9a 29 a0 d0 51 84 22 30 50 ,
98 36 00 c4 8e 21 00 00 1c 52 35 80 80 70 38 1f ,
40 20 20 13 00 c4 8e 21 00 00 1e 01 1d 00 72 51 ,
d0 1e 20 6e 28 55 00 c4 8e 21 00 00 1e 00 00 00 ,
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 71 
```