ValveSoftware/steamlink-sdk #52: Better PS4 Controller support

VanCoding, 2016-12-12:
I was not sure where to ask for this, so I decided to do it here. I'm sorry it it's the wrong repository for this.

I have 3 Dualshock 4 Controllers and a PS4, and recently purchased a SteamLink. I use both the PS and the SL regularly with the same 3 Controllers. But switching them from the PS4 to the SL is currently very inconvenient, while switching them back to the PS is as easy as it can get.

It would be cool if switching a Controller from the PS to the SL would work the same way as switching them back: Plug-in each Controller into the steam link via USB and Press the PS Button, then unplug it and the Bluetooth Connection is established.

Currently, i would not even be able to use my Dualshock Controller, if I only had one. Because if I want to connect a Bluetooth Controller to the SL, i need to do it via the Menu. But to navigate the Menu, i need to plug in the Controler via USB. But when its connected via USB, i cannot put the controller into Bluetooth mode, and thus cannot connect it. So, currently, to switch all my controllers to the steam link, i need to connect one via USB first, put the other 2 into Bluetooth Mode, connect them using the Menu and the first Controller, then unplug the first controller, put it into Bluetoothmode, and then connect it with either the 2nd or 3rd Controller. This makes switching really painful.

Additionally, it would be cool if, as with the PS4, when putting the SteamLink into Stand-by, all connected controllers would automatically turn off. And then when pressing the PS-button, they would automatically revive the SteamLink and connect to it. That doesnt currently work as well.

What do you think?

parkerlreed, 2016-12-19:
The beta update from last Thursday solves some of this https://steamcommunity.com/app/353380/discussions/0/152391995405841458/

> It would be cool if switching a Controller from the PS to the SL would work the same way as switching them back: Plug-in each Controller into the steam link via USB and Press the PS Button, then unplug it and the Bluetooth Connection is established.

When the Steam Link boots up to the pairing screen (for the Steam Controller) and you have already paired the PS4 controller once, just press Share + PS to go into pairing mode and it automatically reconnects to the Link. No need to get back to the pairing menu or to use another controller.

> Additionally, it would be cool if, as with the PS4, when putting the SteamLink into Stand-by, all connected controllers would automatically turn off. And then when pressing the PS-button, they would automatically revive the SteamLink and connect to it. That doesnt currently work as well.

Controllers turn off with the Steam Link but yes the PS4 controlelr is unable to wake it from sleep. I think this is just a limitation of the Steam Link at the moment. The only thing it listens for in sleep mode is the built-in "dongle" and Steam Controllers. It doesn't leave Bluetooth active from what I can see.

parkerlreed, 2016-12-20:
Also 574 just made it so all you have to do is plug the PS4 controller in and it pairs.

> Automatically pair with the PS4 controller when it's plugged in via USB

https://steamcommunity.com/app/353380/discussions/0/152391995405841458/

VanCoding, 2016-12-20:
Wow, that's awesome. I'll try it out when I'm home :)

closed 2016-12-25 by slouken