ValveSoftware/steamlink-sdk #54: caveexpress build fails

parkerlreed, 2016-12-20:
````
[ 95%] Building CXX object src/caveexpress/tests/CMakeFiles/tests_caveexpress.dir/__/__/tests/NoNetworkTest.cpp.o
[ 95%] Linking CXX executable /home/parker/build/steamlink-sdk/examples/caveexpress/caveexpress-src/tests_caveexpress
[ 95%] Built target tests_caveexpress
Scanning dependencies of target soundmapper
[ 96%] Building CXX object src/caveexpress/tools/CMakeFiles/soundmapper.dir/SoundMapper.cpp.o
[ 96%] Linking CXX executable /home/parker/build/steamlink-sdk/examples/caveexpress/caveexpress-src/soundmapper
[ 96%] Built target soundmapper
Scanning dependencies of target jsonconvert
[ 96%] Building CXX object src/caveexpress/tools/CMakeFiles/jsonconvert.dir/JSONConvert.cpp.o
[ 96%] Linking CXX executable /home/parker/build/steamlink-sdk/examples/caveexpress/caveexpress-src/jsonconvert
[ 96%] Built target jsonconvert
Scanning dependencies of target tests
[ 96%] Building CXX object src/tests/CMakeFiles/tests.dir/ByteStreamTest.cpp.o
[ 97%] Building CXX object src/tests/CMakeFiles/tests.dir/EnumTest.cpp.o
[ 97%] Building CXX object src/tests/CMakeFiles/tests.dir/FileTest.cpp.o
[ 97%] Building CXX object src/tests/CMakeFiles/tests.dir/GeneralTest.cpp.o
[ 97%] Building CXX object src/tests/CMakeFiles/tests.dir/LUATest.cpp.o
[ 98%] Building CXX object src/tests/CMakeFiles/tests.dir/MemoryAllocatorTest.cpp.o
[ 98%] Building CXX object src/tests/CMakeFiles/tests.dir/NetworkTest.cpp.o
[ 98%] Building CXX object src/tests/CMakeFiles/tests.dir/NoNetworkTest.cpp.o
[ 98%] Building CXX object src/tests/CMakeFiles/tests.dir/ParticleTest.cpp.o
[ 99%] Building CXX object src/tests/CMakeFiles/tests.dir/SQLiteTest.cpp.o
[ 99%] Building CXX object src/tests/CMakeFiles/tests.dir/SystemTest.cpp.o
[ 99%] Building CXX object src/tests/CMakeFiles/tests.dir/TestAll.cpp.o
[ 99%] Building CXX object src/tests/CMakeFiles/tests.dir/TestGame.cpp.o
[100%] Building CXX object src/tests/CMakeFiles/tests.dir/TestShared.cpp.o
[100%] Building CXX object src/tests/CMakeFiles/tests.dir/TimeManagerTest.cpp.o
[100%] Building CXX object src/tests/CMakeFiles/tests.dir/UITest.cpp.o
[100%] Linking CXX executable /home/parker/build/steamlink-sdk/examples/caveexpress/caveexpress-src/tests
[100%] Built target tests
make: *** No rule to make target 'install/strip'.  Stop.
~/build/steamlink-sdk/examples/caveexpress
cp: cannot stat 'cp-build-steamlink/steamlink/*': No such file or directory
````

jonesmz, 2016-12-20:
Duplicate of https://github.com/ValveSoftware/steamlink-sdk/issues/29

closed 2016-12-20 by slouken

parkerlreed, 2016-12-20:
8 issues open and I failed to see that one, thanks!