ValveSoftware/steamlink-sdk #58: Current Kodi issues

parkerlreed, 2016-12-27:
Just listing these as I come across them

* Starting a video in the YouTube addon for example plays fine but when pausing and resuming, there is quite a noticeable jitter that occurs that doesn't seem to clear up.

* With a custom underscan set on the Steam Link in Settings > Display, Kodi honors this on launch but when you start and then stop a video Kodi starts to ignore the underscan and just takes up the full space

* Steam Controller seems to go completely undetected. I checked Settings > System > (Expert) Input in Kodi and it doesn't list the controller (And the Kodi button remapper doesn't respond to controller presses)

roughnecks, 2017-01-03:
I thought about writing my issue with Kodi Krypton here as well. Let me know if you want me to open another thread.

My keyboard isn't working properly (since Jarvis).
Make/Model: Logitech K400r, Wireless Touch: http://www.logitech.com/it-it/product/wireless-touch-keyboard-k400r
Regional Settings in Kodi: Country: Italy, Language: Italian, Layout: QWERTY Italy, Default Character Set

I cannot type anything except for numbers which show the following characters, going from 1 to 0 (i.e. if I type "1" I get a dot, "2" gets me "A" and so forth):

. A D G J M P T W space

Next two keys (after "zero") on my keyboard are "?" and "^" and in Kodi they work as VOL DOWN and VOL UP respectively.

ImpossivelPT, 2017-01-03:
On Krypton I got the sound out of sync when playing videos, for example on youtube addon

jsalomon617, 2017-01-04:
@roughnecks I've experienced the same thing (Keyboard = Logitech K120, Region/Charset = English (US) QWERTY).  After playing around with it a bit, it seems that it's actually acting as a telephone keypad - hitting 2 repeatedly will cycle through abc2ABC, 3 will yield def3DEF, and so on through the rest of the numbers.

roughnecks, 2017-01-04:
@jsalomon617 Oh, I see, will take another look at it later on. Do other keyboards work for you? I haven't tried with anything else so far... And whose fault is it? Kodi itself?

jsalomon617, 2017-01-04:
@roughnecks I haven't tried any other keyboards either.  I think it's a weird artifact of the native app, since the keyboard works fine otherwise - most relevantly when running Kodi on my [Windows] desktop and streaming via the Link (with the keyboard plugged directly into the Link).

roughnecks, 2017-01-04:
I got another issue: when launching YouTube App, Kodi doesn't fit the screen anymore. Writings on the left side have first letters cut off and the clock on the top right border is cut off too (can't see all of the digits).

parkerlreed, 2017-01-04:
@roughnecks I think that's the same as my second bullet point. It just ignores understand when starting a video.

roughnecks, 2017-01-04:
@parkerlreed Yes, I think you're right, sorry about that - I wasn't sure about "underscan" and what it means exactly (English is not my primary language and I don't understand some technical terms).

llothos, 2017-01-05:
Both kodi 16 and 17 both lock up with keyboard mouse (Rii i8+),  Xbox 360 wireless and wired Xbox one controller.

thecosmicslug, 2017-01-07:
sorry to jump in here, but how can I compile the tvheadend pvr plugin to run on steamlink? is there some way we can compile all the addons?

parkerlreed, 2017-01-07:
There were some mentions of binary addon fixes but I haven't seen how to compile them.

thecosmicslug, 2017-01-08:
I'm guessing the build script needs to be modified to pull the addon code from https://github.com/kodi-pvr/pvr.hts but im not sure how to edit it to pull/compile it at the same time as the rest of the kodi build. 

Wish I knew how to help guys!

MalhavocTheLich, 2017-01-08:
A couple of issues I have had. Version is last Krypton build.

The PVR IPTV Simple Client is missing (usually is already installed, it just needs to be activated. It's quite an important plugin, since it lets you watch internet TVs)

I have a 5.1 audio system, and there's not way it works correctly with this build. If I set it to 2.0, that's fine, but when I enable 5.1 only the front right (!!!) speaker works

parkerlreed, 2017-01-13:
With the new default mapping (and even manually before), Kodi seems to ignore one of the joystick directions when starting up. It may be down, it may be right, but it's always just one.

garbear, 2017-02-10:
I've uploaded a build with most of these issues fixed. I listed the remaining ones in the release notes. https://github.com/garbear/kodi-steamlink/releases/tag/v17.1-rc1-2016-02-07

parkerlreed, 2017-02-10:
Beta build 600?

garbear, 2017-02-10:
Technically the PulseAudio issue was fixed in build 599, but 600 is the first working released build, so yes you'll need to upgrade

parkerlreed, 2017-02-10:
Oh I didn't realize it had jumped up. Thanks!

VenkarD, 2017-03-30:
I noticed such problem : 
1.application crash after a certain period of time,
2.smb large delays
3.as well as when using the iptv plugin has audio but no vidio
P.S
I don't know if this should work :)

msteinbeck, 2017-03-30:
Are you using the stock system of Steam Link or the beta?

VenkarD, 2017-03-30:
I use beta 609