ValveSoftware/steamlink-sdk #60: SDL applications ignore audio output preference

parkerlreed, 2016-12-30:
I just now noticed that when I have a Bluetooth headset connected, SDL applications still play out of HDMI. Only the launcher interface and in-game route correctly

From what I recall this worked correctly with Pulseaudio but since everything has moved to ALSA, it seems to not work anymore.

slouken, 2016-12-30:
The automatic hotplug switching isn't implemented in SDL, but specifying USB and Bluetooth in the audio settings should work in the next beta build.

parkerlreed, 2016-12-30:
Thanks. I'm not hotplugging but the audio just isn't going to the expected source. Bluetooth connection stays active but audio just goes out HDMI. I'll see if that clears up in the update.

parkerlreed, 2017-01-02:
Fixed as of https://github.com/ValveSoftware/steamlink-sdk/commit/57aacbe7654b5e385bed00a679bb2f5cea2eeb74

Thanks @slouken! Also neat to know I can throw libs in that directory to test.

closed 2017-01-02 by parkerlreed

slouken, 2017-01-02:
Great, I'm glad it works. :)