ValveSoftware/steamlink-sdk #62: Native app resolution is 720p instead of 1080p

vially, 2016-12-30:
It seems that most native apps run at 1280x720 resolution.

The easiest way to reproduce this is by installing the latest Kodi build. The `Settings -> System Information -> Video` resolution reported by Kodi is `1280x720@59.94Hz` instead of the expected `1920x1080`.

I've also modified the SDL example by changing the window width/height to:

```c
#define WINDOW_WIDTH    1920
#define WINDOW_HEIGHT   1080
```

When running this modified SDL example it seems that some top and right portions are cut off. Setting the SDL window width/height to 1280x720 seems to fix the issue.

### Setup

#### Public Build 566

`Steam Link -> Settings -> Display` reports a `1920x1080 59.94Hz` resolution

I've tried connecting the steam link to a 1080p TV using the `HDMI->HDMI` cable and also to a 1080p monitor using a `HDMI->DVI` cable and the result is always the same.

### Logs (/mnt/scratch/log/S99steam)

```
[     1] mvSetDisplayOption: set EGL_MVGFX_W*EGL_MVGFX_H from 1280*720 to 1280*720
linux_version_check - failed to open /proc/mv88de3010
linux_version_check - app expects linux kernel revision 6154, please manually check current revision
<<<<< CPU-1 OSAL Init
memory map size for cache:0x0C000000, fd_cache:15, ret:0
memory map size for non-cache:0x02000000, fd_noncache:8
MV_SHM_Init OK:
base_virtaddr_cache:0xAA9E6000, size:0x0C000000
base_virtaddr_noncache:0xA89E6000, size:0x02000000
DBG HEALTH {line: 280 (MV_CC_RPCClnt_Call)} 0x7 
[     2] mvSetDisplayOption: set EGL_MVGFX_AUTOSCALING from 1 to 1
[     3] mvGetDisplay: enter
[     4] mvLoadOptionsFromEnv: user will set these options, return directly
[     5] mvInitDisplay: output resolution 26, 1920*1080
[     6] mvEGL ch3 autoscale (0,0):1280x720 to (0,0):1920x1080
DBG HEALTH {line: 280 (MV_CC_RPCClnt_Call)} 0x7 
[     7] mvInitDisplay: creating 3 fbs, width=1280, height=720, stride = 5120
[     8] mvInitDisplay: success to create 0 th frame buffer, surface=0x113d, size=3686400
[     9] mvInitDisplay: success to create 1 th frame buffer, surface=0x213d, size=3686400
[    10] mvInitDisplay: success to create 2 th frame buffer, surface=0x313d, size=3686400
[    11] mvInitDisplay: successfully
[    12] mvGetDisplay: successfully
[    13] gcoOS_GetDisplayByIndex: set display 0xb701bc68's multiBuffer = 3
libudev: udev_has_devtmpfs: name_to_handle_at on /dev: Function not implemented
[    14] mvExitDisplay enter, is_mv_shm_init=1
Wired controller hung, attempting USB device reset on /dev/bus/usb/001/003
Wired controller hung, attempting USB device reset on /dev/bus/usb/001/003
[    15] mvExitDisplay: exit
```

slouken, 2016-12-30:
Most applications run into memory bandwidth and performance issues on the Steam Link when trying to run at 1080p.

You can use the full resolution of the display by adding this call before initializing SDL video:
SDL_SetHint("SDL_PE_GFX_RESOLUTION", "auto");

vially, 2016-12-30:
Awesome. That fixed the issue. I'll close this now but I think that should be mentioned somewhere in the documentation.

closed 2016-12-30 by vially

retuxx, 2016-12-30:
@vially, does Kodi run smooth with `SDL_SetHint("SDL_PE_GFX_RESOLUTION", "auto");`? If yes, it could be the default configuration couldn't it?

vially, 2016-12-30:
@retuxx I've just patched and rebuilt Kodi using that SDL hint and now it does run at 1920x1080 resolution, but it doesn't seem to run very smoothly. I've only tested navigating the menus (I don't have any videos in my library right now to test them). 

So I either did something wrong or maybe @slouken was right and the hardware is just not enough for some application to be run at the full resolution.

retuxx, 2016-12-30:
@vially, The navigation menus usually run quite smooth. However, there is a system info tab (if I'm right you will find it in settings menu) giving your information about the CPU and memory usage while running Kodi.

rafalJrafal, 2017-05-23:
@vially , I'd like to ask if you could share the binary with higher resolution? that would be helpful

vially, 2017-07-27:
@rafalJrafal Sorry but unfortunately I no longer have that binary available.