ValveSoftware/steamlink-sdk #64: Kodi compile fails

MartenvanWezel, 2016-12-31:
- Ubuntu 16.04.1 LTS
- Installed all the prerequisites listed in README.md

Quick note: I had to restart the build script MANY times - probably because the kodi.tv server kept on refusing to send the source tarballs about 50% of the time. But after that was fixed:

~/steamlink-kodi/kodi-src/tools/depends ~/steamlink-kodi
make: Entering directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/alsa-lib'
make -C armv7a-cros-linux-gnueabi/src
make[1]: Entering directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/alsa-lib/armv7a-cros-linux-gnueabi/src'
make[1]: *** No targets specified and no makefile found. Stop.
make[1]: Leaving directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/alsa-lib/armv7a-cros-linux-gnueabi/src'
Makefile:46: recipe for target 'armv7a-cros-linux-gnueabi/src/.libs/libasound.so' failed
make: *** [armv7a-cros-linux-gnueabi/src/.libs/libasound.so] Error 2
make: Leaving directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/alsa-lib'
make: Entering directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/libsdl2'
make -C armv7a-cros-linux-gnueabi
make[1]: Entering directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/libsdl2/armv7a-cros-linux-gnueabi'
make[1]: *** No targets specified and no makefile found. Stop.
make[1]: Leaving directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/libsdl2/armv7a-cros-linux-gnueabi'
Makefile:30: recipe for target 'armv7a-cros-linux-gnueabi/build/.libs/libSDL2.a' failed
make: *** [armv7a-cros-linux-gnueabi/build/.libs/libSDL2.a] Error 2
make: Leaving directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/libsdl2'
make: *** /home/marten/steamlink-kodi/kodi-src/tools/depends/target/libogg: No such file or directory. Stop.
touch: cannot touch '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/libogg/.installed-armv7a-cros-linux-gnueabi': No such file or directory
make: *** /home/marten/steamlink-kodi/kodi-src/tools/depends/target/libvorbis: No such file or directory. Stop.
touch: cannot touch '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/libvorbis/.installed-armv7a-cros-linux-gnueabi': No such file or directory
~/steamlink-kodi/kodi-src/tools/depends/target/libgpg-error ~/steamlink-kodi/kodi-src/tools/depends ~/steamlink-kodi
make: 'armv7a-cros-linux-gnueabi' is up to date.
grep: armv7a-cros-linux-gnueabi/src/Makefile: No such file or directory
sed: can't read armv7a-cros-linux-gnueabi/src/Makefile: No such file or directory
~/steamlink-kodi/kodi-src/tools/depends ~/steamlink-kodi
make: Entering directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target'
Makefile:153: warning: overriding recipe for target 'linux-system-libs'
Makefile:114: warning: ignoring old recipe for target 'linux-system-libs'
make -C pcre
make[1]: Entering directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/pcre'
make -C armv7a-cros-linux-gnueabi
make[2]: Entering directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/pcre/armv7a-cros-linux-gnueabi'
make[2]: *** No targets specified and no makefile found. Stop.
make[2]: Leaving directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/pcre/armv7a-cros-linux-gnueabi'
Makefile:34: recipe for target 'armv7a-cros-linux-gnueabi/.libs/libpcre.a' failed
make[1]: *** [armv7a-cros-linux-gnueabi/.libs/libpcre.a] Error 2
make[1]: Leaving directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/pcre'
Makefile:114: recipe for target 'pcre' failed
make: *** [pcre] Error 2
make: Leaving directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target'

slouken, 2016-12-31:
Try removing the kodi-src directory and trying again now that the dependencies have been downloaded?

MartenvanWezel, 2017-01-01:
Quick update: Removed kodi-src, retried. Fail. :(

Sadly the output of my build script is lost from my scrollback buffer. It said something like 'the C compiler cannot create an executable'. Will try to repro. I do have this though (referred to from the build script error)

~/steamlink-kodi/kodi-src/tools/depends/target/pcre/armv7a-cros-linux-gnueabi/config.log contains:

configure:3789: /home/toolchain/bin/ --version >&5
./configure: line 3791: /home/toolchain/bin/: No such file or directory
configure:3800: $? = 127
configure:3789: /home/toolchain/bin/ -v >&5
./configure: line 3791: /home/toolchain/bin/: No such file or directory
configure:3800: $? = 127
configure:3789: /home/toolchain/bin/ -V >&5
./configure: line 3791: /home/toolchain/bin/: No such file or directory
configure:3800: $? = 127
configure:3789: /home/toolchain/bin/ -qversion >&5
./configure: line 3791: /home/toolchain/bin/: No such file or directory
configure:3800: $? = 127
configure:3820: checking whether the C compiler works
configure:3842: /home/toolchain/bin/   -g -D_DEBUG  -isystem /home/kodi-deps/armv7a-cros-linux-gnueabi/include    -g -D_DEBUG  -isystem /home/kodi-deps/armv7a-cros-linux-gnueabi/include  -L/home/kodi-deps/armv7a-cros-linux-gnueabi/lib -L/home/kodi-deps/armv7a-cros-linux-gnueabi/lib64     conftest.c  >&5
./configure: line 3844: /home/toolchain/bin/: No such file or directory
configure:3846: $? = 127
configure:3884: result: no
configure: failed program was:

..

Clearly it shouldn't try to create /home/toolchain ? Also when rerunning the build script (after deleting kodi-src again) I still saw some references to /home/kodi-deps - curious.

Will report back in after I do a full wipe&restart. Thanks for working on this, I think kodi would be a big value-add for the steam link!

MartenvanWezel, 2017-01-01:
Here's the output of re-deleting kodi-src and rebuilding.
I'll now remove my deps /home/kodi-deps dir and see if I cn start afresh. See where I strand.

configure.in:52: warning: AC_LANG_CONFTEST: no AC_LANG_SOURCE call detected in body
../../lib/autoconf/lang.m4:193: AC_LANG_CONFTEST is expanded from...
../../lib/autoconf/general.m4:2661: _AC_LINK_IFELSE is expanded from...
../../lib/autoconf/general.m4:2678: AC_LINK_IFELSE is expanded from...
../../lib/m4sugar/m4sh.m4:639: AS_IF is expanded from...
../../lib/autoconf/general.m4:2031: AC_CACHE_VAL is expanded from...
../../lib/autoconf/general.m4:2052: AC_CACHE_CHECK is expanded from...
m4/attributes.m4:87: CC_CHECK_LDFLAGS is expanded from...
m4/attributes.m4:104: CC_NOUNDEFINED is expanded from...
configure.in:52: the top level
configure.in:15: warning: AM_INIT_AUTOMAKE: two- and three-arguments forms are deprecated.  For more info, see:
configure.in:15: http://www.gnu.org/software/automake/manual/automake.html#Modernize-AM_005fINIT_005fAUTOMAKE-invocation
configure.in:44: installing './compile'
configure.in:15: installing './missing'
Makefile.am:20: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
alsalisp/Makefile.am:8: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
alsalisp/Makefile.am: installing './depcomp'
automake: warning: autoconf input should be named 'configure.ac', not 'configure.in'
aserver/Makefile.am:8: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
doc/Makefile.am:5: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
include/Makefile.am:86: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
modules/mixer/simple/Makefile.am:6: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/Makefile.am:86: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/alisp/Makefile.am:11: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/control/Makefile.am:17: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/hwdep/Makefile.am:8: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/mixer/Makefile.am:10: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/pcm/Makefile.am:117: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/rawmidi/Makefile.am:12: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/seq/Makefile.am:13: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/timer/Makefile.am:9: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
test/Makefile.am:22: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
parallel-tests: installing './test-driver'
utils/Makefile.am:13: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
autoreconf: Leaving directory `.'
configure: loading site script /home/kodi-deps/armv7a-cros-linux-gnueabi/share/config.site
checking build system type... x86_64-unknown-linux-gnu
checking host system type... armv7a-cros-linux-gnueabi
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for strip... /usr/bin/strip
configure: WARNING: using cross tools not prefixed with host triplet
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for cross-compiler... /home/toolchain/bin/
checking for gcc... /home/toolchain/bin/
checking whether the C compiler works... no
configure: error: in `/home/marten/steamlink-kodi/kodi-src/tools/depends/target/alsa-lib/armv7a-cros-linux-gnueabi':
configure: error: C compiler cannot create executables
See `config.log' for more details
Makefile:34: recipe for target 'armv7a-cros-linux-gnueabi' failed
make: *** [armv7a-cros-linux-gnueabi] Error 77
make: Leaving directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/alsa-lib'
make: Entering directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/libsdl2'
rm -rf armv7a-cros-linux-gnueabi; mkdir -p armv7a-cros-linux-gnueabi
cd armv7a-cros-linux-gnueabi; /bin/tar --strip-components=1 -xf /home/kodi-deps/xbmc-tarballs/SDL2-2.0.3.tar.gz
cd armv7a-cros-linux-gnueabi; ./autogen.sh
Generating build information using autoconf
This may take a while ...
Now you are ready to run ./configure
cd armv7a-cros-linux-gnueabi; ./configure --prefix=/home/kodi-deps/armv7a-cros-linux-gnueabi --disable-video-directfb --without-x --disable-video-x11
configure: loading site script /home/kodi-deps/armv7a-cros-linux-gnueabi/share/config.site
checking build system type... x86_64-unknown-linux-gnu
checking host system type... armv7a-cros-linux-gnueabi
[... 54 lines cut ...]
cd armv7a-cros-linux-gnueabi; patch -p0 < ../jit_aarch64.patch
patching file sljit/sljitConfigInternal.h
cd armv7a-cros-linux-gnueabi; cp -f /home/kodi-deps/x86_64-linux-gnu-native/share/automake-1.15/config.sub /home/kodi-deps/x86_64-linux-gnu-native/share/automake-1.15/config.guess .; ./configure --prefix=/home/kodi-deps/armv7a-cros-linux-gnueabi --disable-shared --disable-stack-for-recursion --enable-pcre8 --disable-pcre16 --disable-pcre32 --enable-jit --enable-utf --enable-unicode-properties --enable-newline-is-anycrlf
[... cut ...]
configure: error: in `/home/marten/steamlink-kodi/kodi-src/tools/depends/target/pcre/armv7a-cros-linux-gnueabi':
configure: error: C compiler cannot create executables
See `config.log' for more details
Makefile:27: recipe for target 'armv7a-cros-linux-gnueabi' failed
make[1]: *** [armv7a-cros-linux-gnueabi] Error 77
make[1]: Leaving directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target/pcre'
Makefile:114: recipe for target 'pcre' failed
make: *** [pcre] Error 2
make: Leaving directory '/home/marten/steamlink-kodi/kodi-src/tools/depends/target'

slouken, 2017-01-01:
You need to do the build in-tree, in the SDK examples/kodi directory. The latest version of the kodi build script verifies this.

slouken, 2017-01-01:
Also, don't source setenv.sh before you run the build script.

closed 2017-01-01 by slouken

slouken, 2017-01-01:
The latest version of the Kodi build script should check for the situation you ran into and provide useful diagnostic output.

MartenvanWezel, 2017-01-02:
Success. (almost).

Running ./build_steamlink.sh failed pretty much immediately on line 24. 

I had to change it to a hardcoded path:
git am <~/steamlink-kodi/steamlink-sdk/examples/kodi/kodi.patch || exit 1

After that (and restarting the downloads of all the external sources manymany times) I got a build. Haven't tested it yet.

MartenvanWezel, 2017-01-02:
Success. It worked, after a small installing hiccup. (installing seemed to crash, black screen, an occasional soft 'pop' from the speakers), but after a powercycle everything succeeded.

The app was locked into 1280x720x60Hz resolution (according to system settings) and the system wasn't powerful enough to play back anything at full speed though. Also no support for the steam controller- pity. 

It was a fun experiment but I guess I'll have to wait for it to mature. (my goal actually was to just get it to be a video source, playing back at 23.976p - I've been experiencing some stuttering issues with kodi-ubuntu and wanted to debug.. anyway.. )

Thanks for your help guys!

slouken, 2017-01-02:
It should have been able to play H.264 video at full speed. What videos were you having problems with?

MartenvanWezel, 2017-01-02:
Any that I tried, but in particular indeed a couple of h264 videos. All of them were MKV, all H264, AAC audio or AC3 surround audio.

MartenvanWezel, 2017-01-03:
Quick update: I'm happy to help out and debug these issues btw. Happy to experiment with a few files that work for you, or try to find something that works for me. Let me know.

Note: I was only able to find media files off my network share, I couldn't get kodi to find my USB stick.. where does the steamlink mount such drives? Not in /mnt it seemed. Surely streaming over network doesn't add such a 'CPU load' that everything else started to chug?

retuxx, 2017-01-03:
> Note: I was only able to find media files off my network share, I couldn't get kodi to find my USB stick.. where does the steamlink mount such drives? Not in /mnt it seemed.

Currently, Steam Link does not automount mass storage devices. However, I'm currently working on an automount script utilizing udev. Once it is working I'll publish it at Github.

This is related to: https://github.com/ValveSoftware/steamlink-sdk/issues/68

MartenvanWezel, 2017-01-03:
Cool.

Wrt the performance issues on Kodi - I've got a decent amount of experience with debugging kodi issues, but step one to this usually is access to the logs. Is there any way to pull the logs off the device? I don't suppose there's a usable sshd?

I've noticed a 'kodi log viewer' add-on. Perhaps that's worth trying, but I'm not sure if the hardware init logs can be used.
[How to check kodi log file](https://howtomediacenter.com/en/how-to-check-kodi-log-file/) 

As it stands my loose theory is that somehow my build fails to recognize the steamlink's hardware and is running in 'standard vga mode'- so to speak.

kisak-valve, 2017-01-03:
Hello @MartenvanWezel, instructions on enabling ssh on the steamlink can be found at https://github.com/ValveSoftware/steamlink-sdk#ssh-access.

MartenvanWezel, 2017-01-03:
Nice. Thanks kisak/retuxx/slouken, let me dig into the logs etc, and open a new issue if/when appropriate.