ValveSoftware/steamlink-sdk #72: Please add native Steam Controller support in Kodi@SteamLink

rogalmic, 2017-01-10:
I was able to detect steam controller is connected using Krypton KODI build:
https://steamcommunity.com/app/353380/discussions/1/152393186504700250/
+ latest beta Steam Link firmware.

However, I was not able to set up the controller properly using available profiles. The configuration does not allow all analog stick directions. Also, when playing youtube movie, the elapsed time label shows seeking forward all the time, like if some button was permanently pressed.

kisak-valve, 2017-01-10:
Possible duplicate of #58.

rogalmic, 2017-01-19:
After playing for some time via PC streaming, i went back to native Kodi app and steam controller worked fine. After Steam Link reboot the problems came back, could not use down from stick and youtube player was fastforwarding. 

Seems like some initial controller state problem, maybe this info will help investigate...

garbear, 2017-02-10:
The controller issue has been fixed in Kodi 17.1-RC1. Can you give this release a try? https://github.com/garbear/kodi-steamlink/releases/tag/v17.1-rc1-2016-02-07

rogalmic, 2017-02-10:
Steam Controller works fine with 17.1 build above. Both left stick and right pad (as mouse).

I noticed one unrelated problem - trying to exit or shutdown kodi hangs the system.

garbear, 2017-02-10:
The hang at shutdown is due to memory corruption. Somewhere the main thread hits a decision that gets flipped and sends Kodi into an infinite sleep. I've done a depends build of Kodi on x64 and there's no hang at shutdown, so it is probably due to the GLES code, the EGL code or the Steam Link code.

rogalmic, 2017-02-12:
One more thing I just noticed - KODI shows CPU0 utilization at 100% all the time (i think it was not like that before). The link does not get that hot when working, so this could be just a false-positive case.

Overall - playback works ok with this build. I did not experience any crash when starting playback yet.

roughnecks, 2017-02-17:
> The hang at shutdown is due to memory corruption.

This is main thing keeping me from installing Kodi again. How do you exit when it hangs? Unplug the Link?

rogalmic, 2017-02-17:
Yes, unplug the link from power. However, this happens only when kodi was launched - if i use the link for streaming games, shutdown works fine.

roughnecks, 2017-02-17:
Yeah, I know it's just a Kodi's issue and that's why I'll wait for a fix before installing it again.

garbear, 2017-02-17:
I found a solution to the hang at exit, but unfortunately it breaks controllers! I'll keep looking for a better way to solve this.