ValveSoftware/steamlink-sdk #76: Unable to compile QT example

thecosmicslug, 2017-02-05:
the latest update seems to leave me unable to compile the qt example, at first I thought it was just the project im working on so I checked the example and that was giving me the same error message so im pretty sure it must have been the latest commits as it was working fine a coupe of days ago. here is the message where it fails.

`/steamlink-sdk/toolchain/usr/x86_64-pc-linux-gnu/armv7a-cros-linux-gnueabi/binutils-bin/2.24/ld.bfd.real.elf: /home/danny/steamlink-sdk/toolchain/usr/x86_64-pc-linux-gnu/armv7a-cros-linux-gnueabi/gcc-bin/4.9.x-google/../../../../libexec/gcc/armv7a-cros-linux-gnueabi/4.9.x-google/liblto_plugin.so: error loading plugin: /home/danny/steamlink-sdk/toolchain/usr/x86_64-pc-linux-gnu/armv7a-cros-linux-gnueabi/gcc-bin/4.9.x-google/../../../../libexec/gcc/armv7a-cros-linux-gnueabi/4.9.x-google/liblto_plugin.so: invalid ELF header
collect2.elf: error: ld returned 1 exit status
Makefile:240: recipe for target 'example' failed
make: *** [example] Error 1`

slouken, 2017-02-06:
You need to use Git Large File Support. See the update to the top level README

closed 2017-02-06 by slouken

parkerlreed, 2017-02-06:
For anybody else confused: git-lfs is a separate package you have to install.

Jorys-Paulin, 2018-06-25:
@slouken I've done what you said and what @parkerlreed said, but I'm still getting an error:
```bash

parkerlreed, 2018-06-25:
@Jorys-Paulin You need `build-essential` installed

Jorys-Paulin, 2018-06-26:
@parkerlreed I doesn't installed anything because I thought everything was bundled and I wanted to avoid version incompatibility

Jorys-Paulin, 2018-06-26:
@parkerlreed I installed the package on a fresh install, but got this error:
```bash
jorys@DESKTOP-LLSAPKH:/mnt/e/Projects/steamlink-sdk/examples/linux$ source ../../setenv.sh
jorys@DESKTOP-LLSAPKH:/mnt/e/Projects/steamlink-sdk/examples/linux$ make
armv7a-cros-linux-gnueabi-gcc --sysroot=/mnt/e/Projects/steamlink-sdk/rootfs -marm -mfpu=neon -mfloat-abi=hard  -g  -c -o example.o example.c
/mnt/e/Projects/steamlink-sdk/toolchain/bin/armv7a-cros-linux-gnueabi-gcc: 4: exec: /mnt/e/Projects/steamlink-sdk/toolchain/bin/../usr/x86_64-pc-linux-gnu/armv7a-cros-linux-gnueabi/gcc-bin/4.9.x-google/armv7a-cros-linux-gnueabi-gcc: not found
<builtin>: recipe for target 'example.o' failed
make: *** [example.o] Error 127
```

parkerlreed, 2018-06-28:
@Jorys-Paulin I'm willing to bet it doesn't compile on the Windows subsystem for Linux. Any number of things there could be causing issues.

Jorys-Paulin, 2018-06-28:
@parkerlreed Sad, it is supposed to be a full Linux... I also tried under an ubuntu container running docker on a true VM with success for the Linux example, but not the qt one.