ValveSoftware/steamlink-sdk #77: QT compiled programs not runnning anymore

thecosmicslug, 2017-02-06:
Since the move to qt5.8 I am still unable to get my programs to run. Ive added LFS instructions and got compilation to work again, but the program segfaults on the link. I have tried a clean clone of the sdk, and even a factory reset and check for updates on the link. the error messages seem to be relating to qt5.4 still, is there something I need to change in my project?

`/usr/local/Qt-5.4.1/lib/libQt5Core.so.5: version `Qt_5' not found
/usr/local/Qt-5.4.1/lib/libQt5Gui.so.5: version `Qt_5' not found 
/usr/local/Qt-5.4.1/lib/libQt5Widgets.so.5: version `Qt_5' not found`

 the qt examples only produces this one line.

slouken, 2017-02-06:
You need Steam Link build 594 or newer, which is not available yet.

thecosmicslug, 2017-02-06:
Ah! Glad its nothing my end, thanks for letting me know slouken, Looking forward to playing with the next update!

slouken, 2017-02-07:
Build 597 is now available. :)

closed 2017-02-07 by slouken

parkerlreed, 2017-02-07:
@slouken Home directory IO fix also included?

slouken, 2017-02-07:
Yes, it should work much better now.

parkerlreed, 2017-02-07:
@slouken Woohoo! Just tested with RetroArch and it still hung for a second but I think it's just a zip with a lot of small files. Extracted successfully and worked.