ValveSoftware/steamlink-sdk #80: RetroArch SDL2 input needs work

parkerlreed, 2017-02-07:
As with many of the examples, controller input isn't working fully. 

This is all with the SDL2 input driver

Left/right touchpad aren't recognized at all and dual stage triggers aren't picked up on movement or click.

I know some of this isn't even supported normally but figured I would put this up there for reference.

If I'm looking at this correctly the file is https://github.com/libretro/RetroArch/blob/master/input/drivers_joypad/sdl_joypad.c

@anibarro

Sidenote: posted a build with cores over on the Steam Link subreddit https://www.reddit.com/r/Steam_Link/comments/5sofyh/release_retroarch_for_the_steam_link/

parkerlreed, 2017-02-08:
Another user reported that everything works on the DS4 but not dpad right.

anibarro, 2017-02-08:
I don't own a DS4 nor a steam controller. I just tested it with my DS3 and everything works, but I did notice that I cannot map the d-pad on my retro-nes controller even though it does work in the steamlink dashboard.
I'll try to take a look this weekend, but I have no experience with SDL.

raulgbcr, 2017-03-25:
I have some cheap PS3 like controller, actually the Down, Left, Right (Dpad), L2, R2, and some other buttons are not detected =[

mmatis, 2017-04-02:
I think this is actually an issue with RetroArch's sdl2 joypad driver, not the Link sdk. It does the same thing on a "normal" linux box. I submitted an [issue](https://github.com/libretro/RetroArch/issues/4785) on RA's github repo which contains my findings.