ValveSoftware/steamlink-sdk #87: Steam Link Kodi Error

Gramps, 2017-04-17:
The build seems to compile fine except I do not get the "Build complete!" message.  It just kind of ends and the steamlink folder is there.  But when I try to move it to a USB drive I get the following:
```
Error while copying “audiodecoder.2sf.so.17.2”.
Filesystem does not support symbolic links
```
This happens for a few files that are apparently symbolic links.  Not sure if it is because it is formatted to fat32 or not.  Also not sure if the build isn't complete hence the lack of "Build complete!" at the end.  However, there are no errors that I can see anywhere.

thecosmicslug, 2017-04-19:
can confirm I got this error the last time I tried to compile kodi as well

jonesmz, 2017-05-14:
If you add

export PC_FREETYPE_INCLUDEDIR="${DEPS_INSTALL_PATH}/include/freetype2/"
export FREETYPE2_CFLAGS="-I${DEPS_INSTALL_PATH}/include -I${PC_FREETYPE_INCLUDEDIR}"

prior to the ./configure $STEAMLINK_CONFIGURE_OPTS .......

line, it'll get past this error

jonesmz, 2017-05-14:
Something like this... though, I edited a few lines of the patch, so grain of salt, and all.

diff --git a/examples/kodi/build_steamlink.sh b/examples/kodi/build_steamlink.sh
index 7d1e796944..ca841441f0 100755
--- a/examples/kodi/build_steamlink.sh
+++ b/examples/kodi/build_steamlink.sh
@@ -207,6 +207,8 @@ export CFLAGS CXXFLAGS CPPFLAGS
 export LD AR AS NM STRIP RANLIB OBJDUMP
 export PYTHON_VERSION PYTHON_CPPFLAGS PYTHON_LDFLAGS PYTHON_SITE_PKG PYTHON_NOVERSIONCHECK NATIVE_ROOT
 export UDEV_CFLAGS="-I${DEPS_INSTALL_PATH}/include"
+export PC_FREETYPE_INCLUDEDIR="${DEPS_INSTALL_PATH}/include/freetype2/"
+export FREETYPE2_CFLAGS="-I${DEPS_INSTALL_PATH}/include -I${PC_FREETYPE_INCLUDEDIR}"
 export UDEV_LIBS=-ludev
 export CEC_CFLAGS="-I${DEPS_INSTALL_PATH}/include"
 export CEC_LIBS=-lcec

Gramps, 2017-05-15:
Sounds good. I'll try this when I get back into Linux.  Cheers!

Gramps, 2017-05-16:
Yep, that fixed it, sir!  You should get them to add that as a fix.  Thanks again!

closed 2017-05-16 by Gramps

kisak-valve, 2017-05-16:
Could someone submit a pull request with this adjustment?

Gramps, 2017-05-16:
Will do.  Give me about 30 minutes...

Gramps, 2017-05-16:
Done and submitted!