ValveSoftware/steamlink-sdk #93: Kodi: Invalid use of incomplete type 'class CEGLNativeType'

jonesmz, 2017-05-15:
EGLWrapper.cpp: In function 'bool {anonymous}::CorrectGuess(CEGLNativeType*, const string&)':
EGLWrapper.cpp:66:13: error: invalid use of incomplete type 'class CEGLNativeType'
     if(guess->CheckCompatibility())
             ^
In file included from EGLWrapper.cpp:43:0:
EGLWrapper.h:28:7: error: forward declaration of 'class CEGLNativeType'
 class CEGLNativeType;
       ^
EGLWrapper.cpp:68:34: error: invalid use of incomplete type 'class CEGLNativeType'
       if (implementation == guess->GetNativeName() ||
                                  ^
In file included from EGLWrapper.cpp:43:0:
EGLWrapper.h:28:7: error: forward declaration of 'class CEGLNativeType'
 class CEGLNativeType;
       ^
EGLWrapper.cpp: In member function 'bool CEGLWrapper::Initialize(const string&)':
EGLWrapper.cpp:110:7: error: expected primary-expression before ')' token
       )
       ^
EGLWrapper.cpp:114:18: error: invalid use of incomplete type 'class CEGLNativeType'
     m_nativeTypes->Initialize();
                  ^
In file included from EGLWrapper.cpp:43:0:
EGLWrapper.h:28:7: error: forward declaration of 'class CEGLNativeType'
 class CEGLNativeType;
       ^
EGLWrapper.cpp: In member function 'bool CEGLWrapper::Destroy()':
EGLWrapper.cpp:126:16: error: invalid use of incomplete type 'class CEGLNativeType'
   m_nativeTypes->Destroy();
                ^
In file included from EGLWrapper.cpp:43:0:
EGLWrapper.h:28:7: error: forward declaration of 'class CEGLNativeType'
 class CEGLNativeType;
       ^
EGLWrapper.cpp:128:10: warning: possible problem detected in invocation of delete operator: [-Wdelete-incomplete]
   delete m_nativeTypes;
          ^
EGLWrapper.cpp:128:10: warning: invalid use of incomplete type 'class CEGLNativeType'
In file included from EGLWrapper.cpp:43:0:
EGLWrapper.h:28:7: warning: forward declaration of 'class CEGLNativeType'
 class CEGLNativeType;
       ^
EGLWrapper.cpp:128:10: note: neither the destructor nor the class-specific operator delete will be called, even if they are declared when the class is defined
   delete m_nativeTypes;
          ^
EGLWrapper.cpp: In member function 'std::string CEGLWrapper::GetNativeName()':
EGLWrapper.cpp:136:25: error: invalid use of incomplete type 'class CEGLNativeType'
     return m_nativeTypes->GetNativeName();
                         ^
In file included from EGLWrapper.cpp:43:0:
EGLWrapper.h:28:7: error: forward declaration of 'class CEGLNativeType'
 class CEGLNativeType;
       ^
EGLWrapper.cpp: In member function 'bool CEGLWrapper::CreateNativeDisplay()':
EGLWrapper.cpp:145:23: error: invalid use of incomplete type 'class CEGLNativeType'
   return m_nativeTypes->CreateNativeDisplay();
                       ^
In file included from EGLWrapper.cpp:43:0:
EGLWrapper.h:28:7: error: forward declaration of 'class CEGLNativeType'
 class CEGLNativeType;
       ^
EGLWrapper.cpp: In member function 'bool CEGLWrapper::CreateNativeWindow()':
EGLWrapper.cpp:153:23: error: invalid use of incomplete type 'class CEGLNativeType'
   return m_nativeTypes->CreateNativeWindow();
                       ^
In file included from EGLWrapper.cpp:43:0:
EGLWrapper.h:28:7: error: forward declaration of 'class CEGLNativeType'
 class CEGLNativeType;
       ^
EGLWrapper.cpp: In member function 'void CEGLWrapper::DestroyNativeDisplay()':
[... 158 lines cut ...]
CPP     xbmc/filesystem/VirtualDirectory.o
CPP     xbmc/windows/GUIWindowHome.o
CPP     xbmc/filesystem/XbtDirectory.o
AR      xbmc/input/joysticks/input_joysticks.a
CPP     xbmc/filesystem/XbtFile.o
CPP     xbmc/filesystem/XbtManager.o
CPP     xbmc/filesystem/ZeroconfDirectory.o
CPP     xbmc/filesystem/ZipDirectory.o
CPP     xbmc/filesystem/ZipFile.o
CPP     xbmc/filesystem/ZipManager.o
CPP     xbmc/filesystem/NFSFile.o
CPP     xbmc/filesystem/NFSDirectory.o
CPP     xbmc/windows/GUIWindowLoginScreen.o
CPP     xbmc/filesystem/BlurayDirectory.o
ZipFile.cpp: In member function 'virtual int64_t XFILE::CZipFile::Seek(int64_t, int)':
ZipFile.cpp:199:39: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
           if (Read(buf.get(),iToRead) != iToRead)
                                       ^
ZipFile.cpp:218:38: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (Read(buf.get(), iToRead) != iToRead)
                                      ^
ZipFile.cpp:231:38: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (Read(buf.get(), iToRead) != iToRead)
                                      ^
CPP     xbmc/filesystem/BlurayFile.o

jonesmz, 2017-05-15:
In an attempt to disable the parts of Kodi that use this, I modified the configuration line to this

./configure $STEAMLINK_CONFIGURE_OPTS --prefix=/home/apps/kodi --disable-x11 --disable-gl --disable-vdpau --disable-vaapi --enable-sdl --disable-gles --disable-openmax --disable-dbus --disable-alsa --disable-samba --disable-ssh --enable-nf
s --disable-airplay --disable-airtunes --disable-upnp --disable-avahi --disable-mdnsembedded --disable-non-free --disable-mysql --disable-webserver --disable-optical-drive --disable-libxslt --disable-lirc || exit 4

It didn't change anything

jonesmz, 2017-06-05:
Note:

This is on Gentoo.

It's likely that the kodi build script is relying on a system dependency that isn't specified or checked for, instead of pulling the relevant dependency down inside the build root.

zooxy, 2017-06-07:
Apparently PKG_CHECK_MODULES in configure.ac can't resolve sdl2 and some other libraries.
For whatever reason PKG_CONFIG_LIBDIR doesn't include the *.pc files for sdl2. Modifying PKG_CONFIG_LIBDIR to include said *.pc files fixes the compile errors, but kodi still crashes when launched on the steam link

jonesmz, 2017-06-07:
@zooxy  Could you show me which line you modified and how? I'd like to give it a shot on my system

zooxy, 2017-06-07:
```
#
# Finally build Kodi
# 
source "${MARVELL_SDK_PATH}/setenv.sh" || exit 1
OLD_PKG_CONFIG_LIBDIR=$PKG_CONFIG_LIBDIR
source "${DEPS_CONFIG_SITE}"
export PKG_CONFIG_LIBDIR="${PKG_CONFIG_LIBDIR}:${OLD_PKG_CONFIG_LIBDIR}"
```

in build_steamlink.sh line ~200

jonesmz, 2017-06-07:
@zooxy 

Yep, I was able to compile with your change.

Want me to make a pull request to that effect? Or would you like to?

TeamERHUC, 2017-06-07:
Can you launch kodi on the steam link without crashing?

Feel free to make a pull request - I have no idea how any of this git stuff works

zooxy, 2017-06-07:
Can you launch kodi on the steam link without crashing?

Feel free to make a pull request -  I have no idea how any of this git stuff works

jonesmz, 2017-06-07:
I have not had an opportunity yet. I probably won't be able to try until tomorrow.

I will make a pull request with your changes.

jonesmz, 2017-06-08:
The Kodi that I compiled today using your fix launches without crashing. Nice :-)

pdrobek, 2017-07-18:
Do you know how to patch kodi itself to avoid this issue?

jonesmz, 2017-07-18:
@pdrobek Don't think it's an issue with kodi itself. It's an issue with the build root for the steamlink sdk.

The problem should be fixed in the latest master branch of this repository anyway? Why did you want to modify Kodi itself? Are you still having problems?

closed 2017-07-18 by jonesmz

pdrobek, 2017-07-18:
Yep, im unable to compile whole 17.x tree, now i downgrade to 16.1 and i have to see if this move is correct or not. Some ppl said that i need to upgrade my toolchain (gcc4.94) to at least 5.x

jonesmz, 2017-07-18:
Hrmm, yea sorry I really wouldn't know where to start looking.

Are you running into errors that look substantially similar to what I posted at the start of this issue?

Gcc 4.9.4 *does* have some substantial problems with regards to c++14 compliance. It's possible that they simply are using language features that gcc 4.9.4 doesn't support properly, or at all... You might be out of luck if that's the case.

pdrobek, 2017-07-19:
thx for assistance :-)