ValveSoftware/steamlink-sdk #99: Two wireless Xbox pads are seen as one device

radical-equanimity, 2017-07-02:
Firmware on Xbox One S Bluetooth pad is up to date, as is Steam Link. I pair Xbox One S pad via Bluetooth, it is recognized. I register an Xbox 360 Wireless pad (using a generic wireless dongle), it is recognized. When I start Geometry Wars 3 or Gang Beasts, for example, both controllers can navigate the GUI but only one character is controlled by both devices.

There was one instance where I was able to use both pads to control different characters, but nothing in my setup has been changed and now I can't get it to work again.

kisak-valve, 2017-07-11:
Hello @radical-equanimity, per "Fixed multiple controllers sharing the same XInput slot in some circumstances" in the [2017-07-11](https://steamcommunity.com/groups/SteamClientBeta/announcements/detail/1432558654366914066) steam beta client update, please opt into the steam beta client and retest this issue.

radical-equanimity, 2017-07-15:
Controllers appear separate all the time now but sometimes, in Gang Beasts at least, even though they're separate as shown below, only one controller actually does anything. And it controls both characters.
[image]

slouken, 2017-07-17:
I just started Gang Beasts using the July 13th Steam beta client streaming to the Steam Link, using 2 wireless XBox 360 controllers and both controllers were recognized and controlled different characters.

Can you describe your steps fully so we can try to reproduce them here?