ValveSoftware/steamlink-sdk #100: The Steam Link does not have an active firewall

st34ml1nk, 2017-08-12:
There is no active firewall running on the Steam Link. All network traffic to and from the Steam Link is allowed by default.

Please enable a firewall with some basic rules, and let users configure it further if needed.

parkerlreed, 2017-08-12:
But why? A basic NAT on the main router is sufficient enough to stop anything maliciously going in or out.

st34ml1nk, 2017-08-13:
Don't you use a firewall on your personal computer even if is behind a NAT router?

Not all LAN network environments are secure. There are dorms, fraternities, libraries and similar places where there could be any number of people on the same network. You usually do not want people in such environments to be able to access your devices.

Furthermore, streaming over a non-local network was enabled in build 617:

http://steamcommunity.com/app/353380/discussions/0/1353742967806351457

Changes in build 617:
Added the ability to stream from a specific computer, not on the local network (streaming over the Internet is not officially supported)

More than anything though, I would just like to be able to make MY rules for how MY device can access any network.

closed 2017-12-14 by st34ml1nk