ValveSoftware/steamlink-sdk #101: Default root password is not steamlink123

st34ml1nk, 2017-08-12:
The default root password when enabling and connecting to the Steam Link with ssh is not "steamlink123", it is "steamlin".

Please change this is in the documentation, or change the password.

parkerlreed, 2017-08-12:
https://github.com/ValveSoftware/steamlink-sdk/blob/ce30ba161aa03a4f5564a636ac237f58fd42a68b/rootfs/etc/init.d/startup/S30sshd#L14

It supposed to be 123 but I always used "steamlink" and it worked. I wonder if the script just isn't working.

evanpowers, 2017-11-29:
Busybox's steamlink_defconfig has CONFIG_FEATURE_DEFAULT_PASSWD_ALGO="des", and it's the DES password encryption algorithm that's ignoring all but the first eight characters. Reconfiguring so that CONFIG_FEATURE_DEFAULT_PASSWD_ALGO="sha512" would probably be the best fix; alternatively, S30sshd could be changed to specify a non-DES algorithm. Looks like "passwd -a sha512 root" support is already on, but even "passwd -a md5 root" would be enough.

slouken, 2018-01-10:
The default algorithm has been changed to MD5 in Steam Link build 644.

Thanks!

closed 2018-01-10 by slouken