ValveSoftware/steamlink-sdk #102: Native apps are running as root user

st34ml1nk, 2017-08-12:
Native apps, such as the ones found in the examples folder of the Steam Link SDK, are executed and run by the root user.

This is concerning considering some of example apps require "roms" or "add-ons", which may or may not have been downloaded from questionable sources on the internet.

If it is not needed for some reason, please change it so apps are not running with root user permissions.

parkerlreed, 2017-08-12:
root IS the only user. I assume they need it that way anyways to access some of the hardware of the device (GPU wise)

st34ml1nk, 2017-08-13:
root is not the only user, you can see them all with the command cat /etc/passwd. You can also add more users with the adduser command.

I don't know if the apps could run as a different user, but if they could, they probaby should.

Personally, I would not feel safe installing and running any app if I knew it had full access to all of the device.

closed 2017-12-14 by st34ml1nk