ValveSoftware/steamlink-sdk #110: Only first 8 characters of password are checked

randUsr, 2017-11-05:
Only first 8 chars of password are checked for validity when connecting over ssh.

Reproduce:
1. ssh to steamlink
1. for defaultpassword 'steamlin' is sufficient
1. change password with 'passwd' to a long password over 8 chars. (e.g '123456789')
1. log out
1. login with first 8 characters of new password and any arbitrary characters after that. (e.g '12345678123')

MACVTi, 2017-11-11:
I would take a guess that this is caused by the OS using DES encryption, which only uses the first 8 characters for password encryption. Changing this requires all the passwords to be recreated afaik.

My1, 2017-11-27:
ouch, is bcrypt that much of a problem?

evanpowers, 2017-11-28:
It's probably just a bad Busybox default. Upstream Busybox 1.24 defaults to DES and looking in steamlink_defconfig reveals CONFIG_FEATURE_DEFAULT_PASSWD_ALGO="des", so Valve probably just left the default alone.

The best fix for #101 would probably be to change CONFIG_FEATURE_DEFAULT_PASSWD_ALGO.

Until then, specifying a non-DES algorithm explicitly should allow passwords longer than 8 characters, e.g. "passwd -a md5 root". Haven't actually tried it.

My1, 2017-11-28:
busybox? isnt passwd an integral part of basically every linux-style system?

ezelkow1, 2017-11-28:
busybox is what is used on most embedded systems. It's a single executable that combines most of the standard linux commands in to one single stripped down binary. You save on a lot of space and resources, but lose some functionality with some of the commands since it strips out some of the more esoteric stuff

My1, 2017-11-28:
oh I see. I know busybox from android, but honestly didnt expect it on a steamlink.

Manouchehri, 2017-12-24:
CVE-2017-17878 has been assigned.

Hopefully that will encourage a proper fix..

slouken, 2018-01-10:
The default algorithm has been changed to MD5 in Steam Link build 644.

Thanks!

closed 2018-01-10 by slouken

My1, 2018-01-10:
MD5? is that some kind of joke?

Wordpress got a CVE vor using PHPass (which stacks a lot of MD5s together, shouldnt the steam link at least use bcrypt, the widely accepted standard?

Speeddymon, 2018-01-23:
@My1 MD5 is a hashing algorithm. bcrypt is an encryption algorithm.

Passwords stored in /etc/passwd and /etc/shadow are hashes of the encrypted password. Not the actual password. While MD5 is deprecated due to hash collision vulnerabilities, I believe it is sufficient for providing a valid hash of an already encrypted password, because even if someone could figure out the encrypted password via hash collision, they would still then need to decrypt the password.

Additionally, busybox itself is using MD5 for the default at this time until they are ready to make the switch to SHA256/SHA512.

My1, 2018-01-23:
I think you may wanna check again on bcrypt. bcrypt is BASED on blowfish, an encryption algo but in itself it's a hash.

https://en.wikipedia.org/wiki/Bcrypt

quick info: the password_hash function of PHP introduced in PHP5.5 because apparently many cant properly hash passwords, uses bcrypt as default. and unix itself iirc also uses bcrypt for hashing, at least every hash starting with `$2` is based on that thing.

also hashes of the ENCRYPTED password, WTF? normally any even remotely decent password hash takes a salt, and the password, and throws them together into the hash.

also bcrypt has a workload modifier which can get CRAZY when you set it too high (so that when computers get faster, you can make the hashing take more power thereby making bruteforce a LOT harder.

Speeddymon, 2018-01-23:
My apologies, you are correct, and I misworded when I said "hashes of encrypted password".

https://stackoverflow.com/questions/1561174/sha512-vs-blowfish-and-bcrypt

bcrypt is used in place of a hash algorithm.
md5 itself is a hash algorithm.

bcrypt is "better" for all of the reasons you stated above, however upstream BB has not (as of March 2016) provided a way to use bcrypt, hence md5 is their default and is suitable until such time that they provide better options at compile time.

My1, 2018-01-23:
no problem this blowfish/bcrypt thing can be slightly chaotic to people who dont know about this.

also I dont think bcrypt is just used in place of a hashing algo, and that it's not a hashing algo itself.

one may say that it isnt a hash created from zero because it uses parts of blowfish, but on the otherhand these parts from blowfish form a solid base and bcrypt has basically withstood the test of time ans we are still on pretty low numbers for the work factor (this thing can go from 4 to 32, each increase by one DOUBLES the load, web applications usually never go higher than 12, usually rather more like a 10).

the even better choice would of course be argon2 but that is not really realistic.

I rather wwonder why busybox hasnt included this yet. it has been the only sane thing to use for password hashing for quite a few years. while SHA-X (being at least a part of the SHA2 group) is a lot better at collision checking this stuff is so totally fast it's crazy, well it's designed to be so.
 password hashes on the other hand are designed to be slow as hell, and if possible not easily getting accelerated by a lot (e.g. by the use of a lot of RAM (and even the flimsy 4KiB or so of bcrypt is considered a lot here) you can kick GPUs out of business and by taking a lot more ram (16MB and more) you can also make building hashing clusters a LOT more expensive for the attacker.

That is basically the point. keep the defender's costs (e,g, time loss) reasonable while making it really hard and costly for an attacker.

Manouchehri, 2018-01-24:
Okay, so using MD5 _technically_ is a valid solution for CVE-2017-17878.

That being said, it does feel a bit disingenuous given that both I and @evanpowers (https://github.com/ValveSoftware/steamlink-sdk/issues/101#issuecomment-347744995) have already pointed out that [SHA512 works fine on the Steam Link](https://blogger.davidmanouchehri.com/2017/12/steam-link-security-remotely-insecure.html). I don't think this was the intent, but it was the result. It would be great if Valve had a bug bounty* to improve communication.

_*An unpaid bug bounty is still better than none._

My1, 2018-01-24:
while you may close that CVE, Wordpress on the other hand got its own CVE for using phpass, which makes a hash based on stecked MD5, so better than plain MD5, but not a lot.
https://www.cvedetails.com/cve/CVE-2012-6707/
SHA512 would be at least better than MD5 but still on a marginal scale in my opinion.