ValveSoftware/steamlink-sdk #112: OpenGLES 1.1 headers are missing

elvissteinjr, 2017-11-15:
After adding them from an external source, everything links fine with the libraries which are present and appears to run as it should.

It would be beneficial to have them included with the SDK.

slouken, 2017-12-21:
Where did you get them from? Can you create a pull request?

Thanks!

elvissteinjr, 2018-01-04:
Sorry for the late response.

I originally grabbed the headers from my OpenPandora ARM handheld. Since I'm not 100% sure if there aren't any device specific things in those, I didn't post or mention them.

However, the headers from the khronos website seem to work just as fine, so those could be used: https://www.khronos.org/registry/OpenGL/index_es.php#headers
I did not do extensive testing, but my application runs as expected.

Regarding a pull request, the header files would usually go into "/rootfs/usr/include/GLES", but that's supposed to be a mirror of the steamlink rootfs, right?

slouken, 2018-01-10:
These are in the SDK for the Steam Link build 644, thanks!

closed 2018-01-10 by slouken