ValveSoftware/steamlink-sdk #113: Disable SSH

My1, 2017-11-25:
is there a way to disable SSH again when I dont need it? (important: without resetting everything obviously)

just that it isnt open and nobody can poke around without enabling it again

parkerlreed, 2017-11-27:
````
rm `find /mnt/ -name "enable_ssh.txt"`
````

Only thing that enables it is that file.

My1, 2017-11-27:
does mnt have more than the USB sticks, because if not, you are just removing the file from the stick, and if removing the file from the USB stick is enough it would also be enough to just powercycle it without the stick

parkerlreed, 2017-11-29:
@My1 There's a persistent partition in /mnt where the flash drive configs get copied to. That's where the Link checks on consecutive bootups. Removing that file will make it not enable SSH on the next reboot.

My1, 2017-11-29:
oh I see thanks for the info.

closed 2017-12-21 by slouken