ValveSoftware/steamlink-sdk #117: Any chances to mount nfs/smb shares?

Schnabelriechen, 2017-12-09:
I´m trying to install the nfs-common packet onto the link to mount nfs shares. Any way to do this?

parkerlreed, 2017-12-09:
nfs is already in the mount options 

````
# cat /proc/filesystems | awk '{print $NF}' | sed '/^$/d'
sysfs
rootfs
bdev
proc
cgroup
tmpfs
sockfs
pipefs
anon_inodefs
rpc_pipefs
devpts
ext3
ext2
ext4
squashfs
ramfs
vfat
msdos
unionfs
nfs
cifs
ntfs
fuseblk
fuse
fusectl
yaffs
yaffs2
mqueue
mtd_inodefs
oprofilefs
ubifs
````
Try mounting with `mount` to see if it works.

Schnabelriechen, 2017-12-12:
by trying it, it gives me:

> mount monstanas:/volume1/Downloads /home/apps/retroarch/nfs
mount: mounting monstanas:/volume1/Downloads on /home/apps/retroarch/nfs failed: Connection refused

closed 2017-12-15 by Schnabelriechen