ValveSoftware/steamlink-sdk #118: [Feature Request] Compile Docker for Steam Link

RyanEwen, 2017-12-22:
This might be asking a lot but maybe someone will be interested enough and know how to tackle this..

Basically I would love love love to be able to run docker containers on the Link. Maybe it's not technically possible.. maybe it would be too limited since it's not x86, I do not know.

But since we can't install our own OS it would be amazing to be able to run software made for other OSes within containers. Things like nginx (web server). I would love to run a steam-cache from a steam link itself.

csamsel, 2017-12-22:
> Maybe it's not technically possible..

Certainly it's not. It's way to slow.

unknown321, 2017-12-24:
It is definitely possible, there are some articles mentioning running docker on raspberry pis. See also: https://github.com/umiddelb/armhf/wiki/Installing,-running,-using-docker-on-armhf-(ARMv7)-devices

RyanEwen, 2018-02-18:
Got what I wanted: 

* Use kexec to load a different kernel & OS (see [here](https://github.com/lukas2511/steamlink-sdk) for more info or just extract [this Archlinux for Steam Link release](https://github.com/lukas2511/steamlink-sdk/releases) onto an EXT4 formatted USB stick (root p/w is hunter2, preconfigured for DHCP). 
* Boot into the OS and install Docker. That's it. 

If your Steam Link continues to boot into Steam/Big Picture then you may need to factory reset and keep LAN unplugged until there's no picture output from the Steam Link (that means it got into Arch)