ValveSoftware/steamlink-sdk #119: IPv6 Firewall Needed

Manouchehri, 2017-12-23:
Given that #110 silently truncates passwords to 8 characters, having SSHD bind to all IP addresses isn't a great idea. This isn't much of a risk with most home networks due to IPv4 NAT (https://github.com/ValveSoftware/steamlink-sdk/issues/100#issuecomment-322003318), but those with IPv6 will end up having their Steam Link accessible on the internet if there's no incoming firewall.

https://blogger.davidmanouchehri.com/2017/12/steam-link-security-remotely-insecure.html

We really do need iptables / ip6tables.

```
# Allow SSH only for unique local addresses (fc00::/7) and site-local addresses (fe80::/10).

ip6tables -A INPUT -i eth0,mlan0 -s fc00::/7,fe80::/10 -p tcp -dport 22 -j ACCEPT
ip6tables -A INPUT -j DROP
```

Manouchehri, 2017-12-23:
Temporary workaround:

```sh
sed -i 's/#AddressFamily any/AddressFamily inet/;s/#ListenAddress 0.0.0.0/ListenAddress 0.0.0.0/' /etc/sshd_config
/etc/init.d/reboot.sh
```

Manouchehri, 2018-01-04:
CVE-2017-17877 has been assigned.

https://nvd.nist.gov/vuln/detail/CVE-2017-17877

slouken, 2018-01-10:
IPv6 support has been removed in build 644, thanks!

closed 2018-01-10 by slouken

Manouchehri, 2018-01-11:
Seriously? Why remove features instead of fixing them easily?