ValveSoftware/steamlink-sdk #124: Slackware64 14.2 - I Can't compile it from SlackBuild (sbo)

inukaze, 2018-01-26:
Hi there i am trying to build RetroArch 1.7.0 (Lastest Version) but ever show me this message
```
Checking operating system ... Linux 
Checking for suitable working C compiler ... /usr/bin/clang does not work
Error: Cannot proceed without a working C compiler.
Total build time for package RetroArch : 42.53 Sec

```

clang is working normally. is just a symbolic link to clang-3.8 in the same folder.

parkerlreed, 2018-01-26:
Is clang installed?

inukaze, 2018-01-26:
Yes

parkerlreed, 2018-01-26:
Wait... it shouldn't even be using clang. 

https://github.com/ValveSoftware/steamlink-sdk/blob/master/examples/retroarch/build_steamlink.sh#L16

It's sets the C compiler to the included ARM GCC. Something might be wrong with how the script is being sourced.