ValveSoftware/steamlink-sdk #125: Kodi script

Luk164, 2018-01-30:
Kodi script does not currently work on debian. (Just wanted to inform you)

parkerlreed, 2018-01-31:
What doesn't work? Output would be helpful.

Luk164, 2018-01-31:
Here you go

````
~/Projects/Steam_link/steamlink-sdk/examples/kodi/kodi-src/tools/depends ~/Projects/Steam_link/steamlink-sdk/examples/kodi
make: Entering directory '/root/Projects/Steam_link/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native'
make -C m4-native
make[1]: Entering directory '/root/Projects/Steam_link/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/m4-native'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/root/Projects/Steam_link/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/m4-native'
make -C gettext-native
make[1]: Entering directory '/root/Projects/Steam_link/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/gettext-native'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/root/Projects/Steam_link/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/gettext-native'
make -C autoconf-native
make[1]: Entering directory '/root/Projects/Steam_link/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/autoconf-native'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/root/Projects/Steam_link/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/autoconf-native'
make -C automake-native
make[1]: Entering directory '/root/Projects/Steam_link/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/automake-native'
make -C armv7l-linux-native
make[2]: Entering directory '/root/Projects/Steam_link/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/automake-native/armv7l-linux-native'
  GEN      doc/automake-1.15.1
help2man: can't get `--help' info from automake-1.15
Try `--no-discard-stderr' if option outputs to stderr
Makefile:3687: recipe for target 'doc/automake-1.15.1' failed
make[2]: *** [doc/automake-1.15.1] Error 255
make[2]: Leaving directory '/root/Projects/Steam_link/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/automake-native/armv7l-linux-native'
Makefile:29: recipe for target 'armv7l-linux-native/automake' failed
make[1]: *** [armv7l-linux-native/automake] Error 2
make[1]: Leaving directory '/root/Projects/Steam_link/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/automake-native'
Makefile:47: recipe for target 'automake-native' failed
make: *** [automake-native] Error 2
make: Leaving directory '/root/Projects/Steam_link/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native'`
```

parkerlreed, 2018-01-31:
````
help2man: can't get `--help' info from automake-1.15
````

Does `automake --help` return anything?

Luk164, 2018-02-01:
Here is the automake --help output

```
automake --help
Usage: /usr/bin/automake [OPTION]... [Makefile]...

Generate Makefile.in for configure from Makefile.am.

Operation modes:
      --help               print this help, then exit
      --version            print version number, then exit
  -v, --verbose            verbosely list files processed
      --no-force           only update Makefile.in's that are out of date
  -W, --warnings=CATEGORY  report the warnings falling in CATEGORY

Dependency tracking:
  -i, --ignore-deps      disable dependency tracking code
      --include-deps     enable dependency tracking code

Flavors:
      --foreign          set strictness to foreign
      --gnits            set strictness to gnits
      --gnu              set strictness to gnu

Library files:
  -a, --add-missing      add missing standard files to package
      --libdir=DIR       set directory storing library files
      --print-libdir     print directory storing library files
  -c, --copy             with -a, copy missing files (default is symlink)
  -f, --force-missing    force update of standard files

Warning categories include:
  gnu                GNU coding standards (default in gnu and gnits modes)
  obsolete           obsolete features or constructions
  override           user redefinitions of Automake rules or variables
  portability        portability issues (default in gnu and gnits modes)
  extra-portability  extra portability issues related to obscure tools
  syntax             dubious syntactic constructs (default)
  unsupported        unsupported or incomplete features (default)
  all                all the warnings
  no-CATEGORY        turn off warnings in CATEGORY
  none               turn off all the warnings
  error              treat warnings as errors

Files automatically distributed if found (always):
  ABOUT-GNU           INSTALL             config.rpath        ltconfig
  ABOUT-NLS           NEWS                config.sub          ltmain.sh
  AUTHORS             README              depcomp             mdate-sh
  BACKLOG             THANKS              install-sh          missing
  COPYING             TODO                libversion.in       mkinstalldirs
  COPYING.DOC         ar-lib              ltcf-c.sh           py-compile
  COPYING.LESSER      compile             ltcf-cxx.sh         texinfo.tex
  COPYING.LIB         config.guess        ltcf-gcj.sh         ylwrap
  ChangeLog

Files automatically distributed if found (under certain conditions):
  acconfig.h          config.h.bot        configure           configure.in
  aclocal.m4          config.h.top        configure.ac        stamp-vti

Report bugs to <bug-automake@gnu.org>.
GNU Automake home page: <http://www.gnu.org/software/automake/>.
General help using GNU software: <http://www.gnu.org/gethelp/>.

```

parkerlreed, 2018-02-16:
I just realized 1.15 is it's own package. That might be why it bails (if it can't find that version)

https://packages.debian.org/source/stretch/automake-1.15

ilumnatr, 2018-02-18:
I'm running into the same error in Arch.
```

make -C armv7l-linux-native
make[2]: Entering directory '/home/ilumnatr/git/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/automake-native/armv7l-linux-native'
  GEN      doc/automake-1.15.1
help2man: can't get `--help' info from automake-1.15
Try `--no-discard-stderr' if option outputs to stderr
make[2]: *** [Makefile:3687: doc/automake-1.15.1] Error 255
make[2]: Leaving directory '/home/ilumnatr/git/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/automake-native/armv7l-linux-native'
make[1]: *** [Makefile:29: armv7l-linux-native/automake] Error 2
make[1]: Leaving directory '/home/ilumnatr/git/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native/automake-native'
make: *** [Makefile:47: automake-native] Error 2
make: Leaving directory '/home/ilumnatr/git/steamlink-sdk/examples/kodi/kodi-src/tools/depends/native'

```

Available commands for automake:
```
[ilumnatr@alice kodi]$ automake
automake       automake-1.15

```
Also:
```
[ilumnatr@alice kodi]$ automake --version
automake (GNU automake) 1.15.1

```