ValveSoftware/steamlink-sdk #126: [Feature Request] Link streaming does not take over source computer.

eternal-autumn, 2018-02-03:
When streaming through link, no one else can use the computer the game is being streamed on.  It would be nice if someone else could still use the computer while another family member is streaming games through Link to their tv.

parkerlreed, 2018-02-04:
This isn't an issue with the Link. It's inherent to in home streaming. The GPU has to actively be rendering Big Picture/the game in focus. Also the system has no way to handle multiple mice (across all platforms).

DC64, 2018-04-13:
> This isn't an issue with the Link. It's inherent to in home streaming. The GPU has to actively be rendering Big Picture/the game in focus. Also the system has no way to handle multiple mice/input focusing (across all platforms).

Confirmed, if you connect a mouse to the Steam Link, and use the mouse on the local computer at the same time, it becomes a game of tug-o-war.

As of right now, there's no real solution apart from cutting out the Steam Link from the connection and use HDMI to the TV.

*Just thinking here* unless you used an HDMI splitter where the computer and steam link are connected to the inputs, and the TV is the output. I'll have to try that sometime.